torcheeg/torcheeg #1: Slow behaviour reading DEAP Dataset

bruAristimunha, 2022-06-16:
Hello!

Very nice library! Amazing systematic in the emotion's datasets! Thank you so much for your work!

I was trying to use it in a personal project and had one "error." It's taking more than half an hour to consolidate into the intermedium dataset without any offline or online transformation. I am using a reasonable computer, more than 64 RAM with Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz. Do you have any idea why it is so slow?

```
num_worker = 7
dataset = DEAPDataset(io_path=f'./tmp', #If offline, will be save at this path
                      root_path='./data/data_preprocessed_python',
                      label_transform=transforms.Compose([
                          transforms.Select('valence'),
                          transforms.Binary(5.0),
                      ]), num_worker=num_worker)
```

I also had problems with the module torch_geometric, but I think it was just dependency problems.

tczhangzhi, 2022-06-16:
Thanks for your feedback.

Initializing the `DEAPDataset` without any transformation should be complete in under two minutes (in fact, it takes less than a minute on my server).

If the progress bar has not been updated for a long time, it may mean that there is a problem with CPU parallelism, which is the main performance bottleneck at present. Please try setting `num_worker` to `1` and see if the problem still occurs.

If you wait for a long time in `Please wait for the writing process to complete...`, it means that writing data is slow. Please check if your write location involves slow hard drives or network transfers (maybe munge?).

For `torch-geometric`, you may consider reinstalling with the following command:

```
conda install pyg -c pyg
```

You can also refer to the [official documentation](https://pytorch-geometric.readthedocs.io/en/latest/notes/installation.html) and use pip to install:

```
# cpu version
pip install torch-scatter torch-sparse torch-cluster \
torch-spline-conv torch-geometric -f https://data.pyg.org/whl/torch-1.11.0+cpu.html
```

Feel free to ask any questions, we are still in beta, and any feedback is welcome!

bruAristimunha, 2022-06-19:
Hello @tczhangzhi,

Thank you so much for your quick answer.  Your diagnosis of the issue is perfectly accurate. Just the process of writing is really slow. I tested on my notebook instead of my PC and still got the same results. I studied a little the library lmdb, and it seems to me that it is something related to OS or the file system, a problem with the Ubuntu or EXT4.

About the dependency, thanks! Every works perfectly with torch-geo.

closed 2022-06-19 by bruAristimunha

tczhangzhi, 2022-06-20:
Thanks for your kind feedback! And you're absolutely right about `lmdb`, we are optimizing for faster IO speed.