torcheeg/torcheeg #2: Requiring info.csv for SEED dataset

ruyian, 2022-08-23:
Hi there, I am encountering a problem with reading and pre-processing the SEED dataset. 

The info.csv file is required to instantiate the `SEEDDataset` class (otherwise a FileNotFoundError would be raised). It is not generated on its own.

I think I might have missed out on something. Please kindly give me some suggestions on how to instantiate the class.

bruAristimunha, 2022-09-03:
Hello @ruyian, 

I'm not one of the maintainers, but I've had this problem before. I was trying to save the intermediate state to an already existing folder. Solved it when I changed the folder name or deleted the folder. What command are you trying exactly?

tczhangzhi, 2022-09-07:
Sorry for the late reply.

TorchEEG's dataset automatically generates intermediate variable IO (used to speed up subsequent dataset reads) including info.csv based on the original dataset.

Please pass in parameters for `SEEDDataset` as described in the [documentation](https://torcheeg.readthedocs.io/en/latest/torcheeg.datasets.html#seeddataset). A simple example is:

````
dataset = SEEDDataset(io_path=str(TMP_PATH / 'seed'),
                         root_path=str(DATA_PATH),
                         label_transform=transforms.Compose([
                             transforms.Select('emotion'),
                             transforms.Binary(5.0),
                         ]),
                         num_worker=0)
````

If you encounter problems and want to flush the generated intermediate IO, please delete the local IO according to the printed log, for example: `The target folder already exists, if you need to regenerate the database IO, please delete the path /content/drive/MyDrive/work/ tmp_out/seed.`
```
rm -rf /content/drive/MyDrive/work/ tmp_out/seed
```

We provide a minimal use case on google colab:
https://colab.research.google.com/drive/19jAje4CdB_u8tfa9qumQbVsufnFm44OJ?usp=sharing

closed 2023-03-30 by tczhangzhi