torcheeg/torcheeg #10: TypeError: can only concatenate list (not "int") to list

tiensu, 2023-01-12:
I tried torcheeg with SEED dataset. My code as bellow:
`dataset = SEEDDataset(io_path=f'./seed',
                              root_path='../../DATASET/SEED_EEG/Preprocessed_EEG',
                              offline_transform=transforms.Compose([
                                  transforms.BandDifferentialEntropy(),
                                  transforms.ToGrid(SEED_CHANNEL_LOCATION_DICT)
                              ]),
                              online_transform=transforms.ToTensor(),
                              label_transform=transforms.Compose([
                                  transforms.Select(['emotion']),
                                  transforms.Lambda(lambda x: x + 1)
                              ]),
                              num_worker=4)
print(dataset[0])`

But when running the code, I met an error:
`TypeError                                 Traceback (most recent call last)
Cell In[50], line 1
----> 1 print(dataset[0])

File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torcheeg/datasets/module/emotion_recognition/seed.py:184, in SEEDDataset.__getitem__(self, index)
    181     signal = self.online_transform(eeg=eeg)['eeg']
    183 if self.label_transform:
--> 184     label = self.label_transform(y=info)['y']
    186 return signal, label

File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torcheeg/transforms/any/compose.py:54, in Compose.__call__(self, *args, **kwargs)
     51     raise KeyError("Please pass data as named parameters.")
     53 for t in self.transforms:
---> 54     kwargs = t(**kwargs)
     55 return kwargs

File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torcheeg/transforms/any/lambd.py:50, in Lambda.__call__(self, *args, **kwargs)
     43 def __call__(self, *args, **kwargs) -> Dict[str, any]:
     44     r'''
     45     Args:
     46         x (any): The input.
     47     Returns:
     48         any: The transformed output.
     49     '''
...
---> 10                                   transforms.Lambda(lambda x: x + 1)
     11                               ]),
     12                               num_worker=4)

TypeError: can only concatenate list (not "int") to list`

Can anyone help me how to resolve this problem?
Thank you very much!

tczhangzhi, 2023-01-13:
Greeting,

In TorchEEG version 1.0.11, when a list is passed in `transforms.Select()`, TorchEEG returns a list of labels. For example:
```
info = {'valence': 4.5, 'arousal': 5.5, 'subject_id': 7}
transformed_label = Select(key=['valence'])(y=info)
transformed_label['y'] == [4.5]
```
When a string is passed in `transforms.Select()`, TorchEEG returns a label. For example:
```
info = {'valence': 4.5, 'arousal': 5.5, 'subject_id': 7}
transformed_label = Select(key='valence')(y=info)
transformed_label['y'] == 4.5
```
Therefore, you can consider the following modifications:
```
dataset = SEEDDataset(io_path=f'./seed',
                               root_path='../../DATASET/SEED_EEG/Preprocessed_EEG',
                               offline_transform = transforms. Compose([
                                   transforms.BandDifferentialEntropy(),
                                   transforms.ToGrid(SEED_CHANNEL_LOCATION_DICT)
                               ]),
                               online_transform = transforms.ToTensor(),
                               label_transform = transforms. Compose([
                                   transforms. Select('emotion'), # see here!!!!!!!!!!!!!!!!!!
                                   transforms.Lambda(lambda x: x + 1)
                               ]),
                               num_worker=4)
```
You can also iterate over the list in the `Lambda` function if you don't want to modify `Select(['valence'])`.

tczhangzhi, 2023-01-13:
We'll update this problem on the docs, and we can also adopt a PR if you'd like (*^▽^*)

tiensu, 2023-01-16:
Thank you, tczhangzhi! I understood.
Could I add one more question?
After loading the SEED dataset with the above code, I want to train a classifier with a CCNN model.
```
for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    model = CCNN().to(device)
    optimizer = torch.optim.Adam(model.parameters(), lr=1e-4)

    train_loader = DataLoader(train_dataset, batch_size=batch_size, shuffle=True)
    val_loader = DataLoader(val_dataset, batch_size=batch_size, shuffle=True)
    
    epochs = 5
    for t in range(epochs):
        print(f'Epoch {t+1}\n -------------------')
        train(train_loader, model, loss_fn, optimizer)
        valid(val_loader, model, loss_fn)
    print('Done!')
```
but, I met an error:
```
ValueError                                Traceback (most recent call last)
Cell In[18], line 11
      9 for t in range(epochs):
     10     print(f'Epoch {t+1}\n -------------------')
---> 11     train(train_loader, model, loss_fn, optimizer)
     12     valid(val_loader, model, loss_fn)
     13 print('Done!')

Cell In[5], line 17, in train(dataloader, model, loss_fn, optimizer)
     11 # X = batch[0]
     12 # y = torch.from_numpy(np.asarray(batch[1])).to(device)
     13 # y = batch[1]
     14 
     15 # Computer prediction error
     16 pred = model(X)
---> 17 loss = loss_fn(pred, y)
     19 # backpropagation
     20 optimizer.zero_grad()
File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torch/nn/modules/module.py:1194, in Module._call_impl(self, *input, **kwargs)
   1190 # If we don't have any hooks, we want to skip the rest of the logic in
   1191 # this function, and just call forward.
   1192 if not (self._backward_hooks or self._forward_hooks or self._forward_pre_hooks or _global_backward_hooks
   1193         or _global_forward_hooks or _global_forward_pre_hooks):
...
   3024 if size_average is not None or reduce is not None:
   3025     reduction = _Reduction.legacy_get_string(size_average, reduce)
-> 3026 return torch._C._nn.cross_entropy_loss(input, target, weight, _Reduction.get_enum(reduction), ignore_index, label_smoothing)

ValueError: Expected input batch_size (64) to match target batch_size (0).
```

If I changed the loading of the SEED dataset as below:
```
dataset2 = SEEDDataset(io_path=f'./seed2',
                        root_path='../../DATASET/SEED_EEG/Preprocessed_EEG',
                        online_transform=transforms.Compose([
                            transforms.ToTensor(),
                            transforms.To2d()
                        ]),
                        label_transform=transforms.Compose([
                            transforms.Select('emotion'),
                            transforms.Lambda(lambda x: x + 1)
                        ]))
```
The error is:
```
RuntimeError                              Traceback (most recent call last)
Cell In[17], line 11
      9 for t in range(epochs):
     10     print(f'Epoch {t+1}\n -------------------')
---> 11     train(train_loader, model, loss_fn, optimizer)
     12     valid(val_loader, model, loss_fn)
     13 print('Done!')

Cell In[5], line 16, in train(dataloader, model, loss_fn, optimizer)
     10 y = batch[1][0].to(device)
     11 # X = batch[0]
     12 # y = torch.from_numpy(np.asarray(batch[1])).to(device)
     13 # y = batch[1]
     14 
     15 # Computer prediction error
---> 16 pred = model(X)
     17 loss = loss_fn(pred, y)
     19 # backpropagation
File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torch/nn/modules/module.py:1194, in Module._call_impl(self, *input, **kwargs)
   1190 # If we don't have any hooks, we want to skip the rest of the logic in
   1191 # this function, and just call forward.
   1192 if not (self._backward_hooks or self._forward_hooks or self._forward_pre_hooks or _global_backward_hooks
   1193         or _global_forward_hooks or _global_forward_pre_hooks):
...
    458                     _pair(0), self.dilation, self.groups)
--> 459 return F.conv2d(input, weight, bias, self.stride,
    460                 self.padding, self.dilation, self.groups)

RuntimeError: Given groups=1, weight of size [64, 4, 4, 4], expected input[64, 1, 65, 203] to have 4 channels, but got 1 channels instead
```
Can you show me how to overcome these errors?
Thank you very much!

tczhangzhi, 2023-01-16:
For question 1, the error means that the length of `y` is 0, which does not match the length of the predicted result.

The possible cause of this problem is that the variable `y` has not been assigned a value correctly. Did you do something like `y = batch[1]`?

For question 2, these codes work well:
```
dataset2 = SEEDDataset(io_path=f'./seed2',
                        root_path='../../DATASET/SEED_EEG/Preprocessed_EEG',
                        online_transform=transforms.Compose([
                            transforms.ToTensor(),
                            transforms.To2d()
                        ]),
                        label_transform=transforms.Compose([
                            transforms.Select('emotion'),
                            transforms.Lambda(lambda x: x + 1)
                        ]))
```

The only problem you need to know is the shape of the input that the different models accept. For this part, you can refer to the original paper or the TorchEEG document.

For example, the ideal input shape for CCNN is `[n, 4, 9, 9]` (https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.models.CCNN.html#torcheeg.models.CCNN). The ideal input shape for TSCeption is `[n, 1, 28, 512]` (https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.models.TSCeption.html#torcheeg.models.TSCeption). Your preprocessed results should conform to these shapes.

For your `online_transform`, you may want to apply a CNN-based model on raw EEG data, which is the aim of TSCeption. Thus, you can let:

```
model = TSCeption(num_classes=2,
                  num_electrodes=65, # see here [64, 1, *65*, 200]
                  sampling_rate=200, # see here [64, 1, 65, *200*]
                  num_T=15,
                  num_S=15,
                  hid_channels=32,
                  dropout=0.5)
```

tiensu, 2023-01-16:
Thank @tczhangzhi !
I applied your comment, but a new error occurred:
```
RuntimeError                              Traceback (most recent call last)
Cell In[42], line 19
     17 for t in range(epochs):
     18     print(f'Epoch {t+1}\n -------------------')
---> 19     train(train_loader, model, loss_fn, optimizer)
     20     valid(val_loader, model, loss_fn)
     21 print('Done!')

Cell In[22], line 9, in train(dataloader, model, loss_fn, optimizer)
      6 y = batch[1].to(device)
      8 # Computer prediction error
----> 9 pred = model(X)
     10 print(f'Pred: {pred}')
     11 print(f'Target: {y}')

File ~/anaconda3/envs/eeg/lib/python3.8/site-packages/torch/nn/modules/module.py:1194, in Module._call_impl(self, *input, **kwargs)
   1190 # If we don't have any hooks, we want to skip the rest of the logic in
   1191 # this function, and just call forward.
   1192 if not (self._backward_hooks or self._forward_hooks or self._forward_pre_hooks or _global_backward_hooks
   1193         or _global_forward_hooks or _global_forward_pre_hooks):
-> 1194     return forward_call(*input, **kwargs)
   1195 # Do not call functions when jit is used
   1196 full_backward_hooks, non_full_backward_hooks = [], []
...
    458                     _pair(0), self.dilation, self.groups)
--> 459 return F.conv2d(input, weight, bias, self.stride,
    460                 self.padding, self.dilation, self.groups)

RuntimeError: Calculated padded input size per channel: (62 x 52). Kernel size: (65 x 1). Kernel size can't be greater than actual input size
```
Btw, could you provide me a sample from loading the SEED dataset, reshaping, then training with a CNN model, such as CCNN, TSCeption, etc. Actually, I don't have the DEAP dataset. I have only the SEED dataset.  The torcheeg libraries, has only sample code for the DEAP dataset.
```
dataset = DEAPDataset(io_path=f'./deap',
                    root_path='./data_preprocessed_python',
                    offline_transform=transforms.Compose([
                        transforms.BandDifferentialEntropy(),
                        transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT)
                    ]),
                    online_transform=transforms.ToTensor(),
                    label_transform=transforms.Compose([
                        transforms.Select('valence'),
                        transforms.Binary(5.0),
                    ]))
        model = CCNN(num_classes=2, in_channels=4, grid_size=(9, 9))
```

I really appreciate your help!

tczhangzhi, 2023-01-16:
Please refer to the documentation, e.g.,

https://torcheeg.readthedocs.io/en/latest/auto_examples/examples_seed_feature_ada.html#sphx-glr-auto-examples-examples-seed-feature-ada-py

Later today I will upload some new cases in the gallery:

https://torcheeg.readthedocs.io/en/latest/auto_examples/index.html

tiensu, 2023-01-17:
Thank you, @tczhangzhi !
I will try your recommendation!

closed 2023-03-17 by tczhangzhi