torcheeg/torcheeg #12: How to add new models?

tiensu, 2023-01-18:
When I studied papers about EEG analysis, I met many types of models. For example, SincNet-R, DGCNN, DNN, SNN, H-ATT-BGRU, BioCNN, MLF-CapsNet, RACNN, etc.
How can I add those models to the torcheeg library to use?
I am understanding that, I must do code models by myself, then add to ~/torcheeg/models. Is that right?

tczhangzhi, 2023-01-18:
For the question "How can I add those models to the TorchEEG library to use?", please refer to the section "Define the Model" in the documentation:

https://torcheeg.readthedocs.io/en/latest/introduction_by_example.html

For the question, "If there is no matching between the size of data after the transform steps and the expected input size of the model, how to resolve it?", what you need to pay attention to is that most authors prefer to design their own data preprocessing process. You can reproduce the flow stated by the authors by combining `transforms` in TorchEEG to derive a suitable input of the model, which is one of the conveniences provided by TorchEEG.

In fact, TorchEEG has reproduced many models (and many are in the process of being worked on). If you urgently need a new reproduction, you can provide a link to the paper. If it was in our original plan, we will push the implementation earlier.

tiensu, 2023-01-18:
Thank you your answer, @tczhangzhi !
I am planning to implement the following papers:
1. DREAMER dataset
- Liu, Y., Ding, Y., Li, C., Cheng, J., Song, R., Wan, F., & Chen, X. (2020). Multi-channel EEG-based emotion recognition via a multi-level features guided capsule network. Computers in Biology and Medicine, 123, 103927. https://doi.org/10.1016/j.compbiomed.2020.103927
- Cui, H., Liu, A., Zhang, X., Chen, X., Wang, K., & Chen, X. (2020). EEG-based emotion recognition using an end-to-end regional-asymmetric convolutional neural network. Knowledge-Based Systems, 205, 106243. https://doi.org/10.1016/j.knosys.2020.106243
2. SEED & DEAP dataset
- T. Song, W. Zheng, P. Song and Z. Cui, "EEG Emotion Recognition Using Dynamical Graph Convolutional Neural Networks," in IEEE Transactions on Affective Computing, vol. 11, no. 3, pp. 532-541, 1 July-Sept. 2020, doi: 10.1109/TAFFC.2018.2817622.
- Liu, J., Wu, G., Luo, Y., Qiu, S., Yang, S., Li, W., & Bi, Y. (2020). EEG-Based Emotion Classification Using a Deep Neural Network and Sparse Autoencoder. Frontiers in Systems Neuroscience, 14. https://doi.org/10.3389/fnsys.2020.00043
- Y. Luo et al., "EEG-Based Emotion Classification Using Spiking Neural Networks," in IEEE Access, vol. 8, pp. 46007-46016, 2020, doi: 10.1109/ACCESS.2020.2978163.

If any paper is in your plan, please push implementation earlier! The more papers are done by you, the more research time I save.
I really appreciate your help!

tiensu, 2023-01-27:
Hi @tczhangzhi !
How can I receive notification about a new model being added event to torcheeg? I am looking forward to it.

tczhangzhi, 2023-01-27:
Thank you for your attention. We are aligning with the performance reported by the authors.

You can follow the progress of this project via the "watch" project (the button near "star" and "fork"). At the same time, all changes to the project will be reported in the release log: https://github.com/torcheeg/torcheeg/releases

tiensu, 2023-01-27:
Thank you, @tczhangzhi ! I am watching the project now.

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi

savitha-suresh, 2025-01-18:
> Thank you your answer, [@tczhangzhi](https://github.com/tczhangzhi) ! I am planning to implement the following papers:
>
>     1. DREAMER dataset
>
>
>     * Liu, Y., Ding, Y., Li, C., Cheng, J., Song, R., [...]

@tiensu  I am also trying to implement the RACNN. Could you please tell me if you managed to implement it?