torcheeg/torcheeg #14: How to choose the parameters of Transformer model ArjunViT?

ceciliaxu1, 2023-02-21:
Thank you for your excellent  library. I tried to use the Transformer model ArjunViT on my own dataset ，

**Initialization of the model**   
model = ArjunViT(chunk_size=250, #采样频率250
                 t_patch_size=50,
                 hid_channels=64,
                 num_electrodes=64, #电极个数64
                 num_classes=2).to(device)

**Processing of dataset.**
    dataset = NumpyDataset(X=X,
                        y=y,
                        io_path=f'./tmp_out/transformer/deta',
                        offline_transform=transforms.Compose([
                              transforms.MeanStdNormalize(), 
                          ]),
                        online_transform=transforms.ToTensor(),
                        label_transform=transforms.Compose([
                            transforms.Select('label'),
                        ]),
                        num_worker=2,
                        )

**The following error was obtained.**
  File "transformer.py", line 292, in <module>
    train_loss = train(train_loader, model, loss_fn, optimizer)
  File "transformer.py", line 43, in train
    pred = model(X)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/torcheeg/models/transformer/arjun_vit.py", line 151, in forward
    x = self.to_patch_embedding(x)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/container.py", line 204, in forward
    input = module(input)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1190, in _call_impl
    return forward_call(*input, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (2560x50 and 3200x64)

1. I would like to ask how to choose the parameters of the model

1. What is the input data format of the model, is it “channel x time series”?

tczhangzhi, 2023-02-22:
Greeting,

Sorry for the late reply. Your setup looks fine to me.

Depending on your setup:
```
model = ArjunViT(chunk_size=250,
t_patch_size=50,
hid_channels=64,
num_electrodes=64,
num_classes=2).to(device)
```
The elements in the dataset should be of the shape:
```
dataset[0][0].shape
# [64, 250] #num_electrodes * chunk_size
```
It is worth noting that `chunk_size` % `t_patch_size` should be 0. According to your error, I suggest you check whether `num_electrodes` is wrong? Is the sample shape in your dataset `[1, 250]`?
```
dataset = NumpyDataset(
    X=X,
    y=y,
    io_path=f'./tmp_out/transformer/deta',
    offline_transform=transforms.Compose([
        transforms.MeanStdNormalize(),
    ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select('label'),
    ]),
    num_worker=2,
)
```
Your X should have a shape of `[number of samples, 64, 250]`, not `[64, 250]`. For other parameters, I recommend you to read the documentation. https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.models.ArjunViT.html#torcheeg.models.ArjunViT

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi