torcheeg/torcheeg #15: Tried torcheeg with DEAP dataset but the code cannot assign value to the variable ‘dataset’

pentafingerLQCQ, 2023-03-14:
tried torcheeg with DEAP dataset but the code cannot assign value to the variable ‘dataset’

after got outpot as bellow, i met an error : [[name 'dataset' is not defined]]

`

D:\miniconda3\lib\site-packages\pkg_resources\__init__.py:123: PkgResourcesDeprecationWarning: -PKG-VERSION is an invalid version and will not be supported in a future release
  warnings.warn(

[DEAP]:   0%| ____________________________________________________________________________ | 0/32 [00:00<?, ?it/s]

D:\miniconda3\lib\site-packages\pkg_resources\__init__.py:123: PkgResourcesDeprecationWarning: -PKG-VERSION is an invalid version and will not be supported in a future release
  warnings.warn(

[DEAP]:   0%|_____________________________________________________________________________| 0/32 [00:00<?, ?it/s]

`

Can anyone help me how to resolve this problem?
Thank you very much!

pentafingerLQCQ, 2023-03-15:
python --version

Python 3.8.16

python -c "import torch;print(torch.__version__)"

1.11.0

tczhangzhi, 2023-03-15:
I have set up the same environment as you locally and everything works fine. I noticed that multiprocessing doesn't behave as expected on win10, maybe you're running on windows? A quick fix is to set num_worker to 0.

```
dataset = DEAPDataset(io_path=str(TMP_PATH / 'deap'),
                      root_path=str(DATA_PATH),
                      offline_transform=transforms.Compose(
                          [transforms.BandDifferentialEntropy(),
                           transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT)]),
                      online_transform=transforms.Compose([transforms.BaselineRemoval(),
                                                           transforms.ToTensor()]),
                      label_transform=transforms.Compose([
                          transforms.Select('valence'),
                          transforms.Binary(5.0),
                      ]), num_worker=0) # here!
```
Here is a notebook example: https://colab.research.google.com/drive/1VgNT_GiW5J3PYexjpFGdHxCJ2ky_-6jH#scrollTo=4xM2mRwThj3l

pentafingerLQCQ, 2023-03-16:
Thanks for your patience.
It is so weird that the code works well today,though I didn't change anything about my environment or the code itself.It just work suddenly.

Also the warning like ↓↓↓

`:\miniconda3\lib\site-packages\pkg_resources\__init__.py:123: PkgResourcesDeprecationWarning: -PKG-VERSION is an invalid version and will not be supported in a future release`
`warnings.warn(`

still appeared.But it doesn't seems matter.

I will still inform you if anything wrong happen.

tczhangzhi, 2023-03-16:
Glad to hear you're doing well, although we don't know what caused the error. Feel free to contact us if you encounter any problems.

closed 2023-03-17 by tczhangzhi