torcheeg/torcheeg #17: How to solve the problem"Input type (double) and bias type (float) should be the same"about 'trainer.fit()'

pentafingerLQCQ, 2023-03-30:
I tryed to use EEG datasets collected by our team,but I meet this error and didn't know how to solve it.

At this stage, we are not concerned with the details of the deep learning function. We just want to do simple Emotion-recognition tasks through torchEEG. 

First of all, can this problem be solved?Will the code be good after that？Maybe there will still be some error followed?Maybe my code for inputting and transforming data is not written correctly?Where can I find tutorials about using custom datasets on torch?

`Input type (double) and bias type (float) should be the same`
`  File "■■privacy■■\custom_dataset_Torch.py", line 125, in <module>`
 `   trainer.fit(train_loader, val_loader, num_epochs=50)`
`RuntimeError: Input type (double) and bias type (float) should be the same`

tczhangzhi, 2023-03-30:
Can you provide the detailed setup of your dataset?

1. For example, if your data set is composed of European data format (.edf), General data format (.gdf), or Neuroscan CNT (.cnt) format files, you can consider using `mne` to read and then pass in `MNEDataset`.
Related documentation: https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.MNEDataset.html#torcheeg.datasets.MNEDataset

2. If you have read the dataset into and processed it with `Numpy` (data type `np.ndarray`), you can use `NumpyDataset`.
Related documentation: https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.NumpyDataset.html#torcheeg.datasets.NumpyDataset

3. As for the error message you gave, it shows that the data type does not match. You need to convert the data to `float` in the `Dataset` you defined, for example by `some_tensor.float()`. If the data you define is `np.array`, then it needs to be float32 (`some_array.astype (np.float32)`) format.

pentafingerLQCQ, 2023-03-30:
I tryed `NumpyDataset`，it's fine.But here is another error:

about`    trainer.fit(train_loader, val_loader, num_epochs=50)`
`AttributeError: 'dict' object has no attribute 'to'`

Here is a part of my code:

    ds = NumpyDataset(X=xxx,   #(ndarray_like
                       y=yyy,  #{'str':ndarray_like}
                       io_path='tmp_out/custom_dataset',
                       offline_transform=transforms.Compose(
                           [transforms.BandDifferentialEntropy()]),
                       online_transform=transforms.ToTensor(),
                       label_transform=None,
                       num_worker=0,
                       num_samples_per_worker=100)

    k_fold = KFold(n_splits=5, split_path='tmp_out/custom_dataset/split')

    for i, (train_dataset, val_dataset) in enumerate(k_fold.split(ds)):

        model = TSCeption(num_electrodes=14,
                            num_classes=3,
                            num_T=15,
                            num_S=15,
                            in_channels=1,
                            hid_channels=32,
                            sampling_rate=256,
                            dropout=0.5)

        trainer = ClassificationTrainer(model=model,
                                                lr=0.0001,
                                                weight_decay=0.0001,
                                                device_ids=[])

        train_loader = DataLoader(train_dataset,
                                    batch_size=1,
                                    shuffle=True,
                                    num_workers=0)
        val_loader = DataLoader(val_dataset,
                                    batch_size=1,
                                    shuffle=False,
                                    num_workers=0)

        trainer.fit(train_loader, val_loader, num_epochs=50)
        trainer.test(val_loader)
        trainer.save_state_dict(f'./tmp_out/examples_mne_dataset/weight/{i}.pth')

Could you please check what is incorrect?Thank you.

tczhangzhi, 2023-03-30:
Please check your `label_transform`, you should use `transforms.Select('str')` to select the appropriate label from the label ndarray dictionary as in the documentation.
```
# label_transform=None in your codes
...
label_transform=transforms.Compose([
                           transforms.Select('valence'),
                           transforms.Binary(5.0),
                       ]),
...
```
Otherwise, a dictionary will be returned and the trainer has no idea what you are going to use as labels for training.

pentafingerLQCQ, 2023-03-30:
That works.Your suggestion helped a lot, I am more than grateful.

tczhangzhi, 2023-05-18:
Since this question has been inactive for a long time, I've closed this it. If you have further questions, please feel free to reopen this issue.

closed 2023-05-18 by tczhangzhi