torcheeg/torcheeg #20: Folder data and dataframe data.

talhaanwarch, 2023-04-24:
I have a particular scenario, let say we have folders having files where  folder represent the class names. 
I can use NumpyDataset  [NumpyDataset](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.NumpyDataset.html#torcheeg.datasets.NumpyDataset)or may be [MNEDataset](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.MNEDataset.html#torcheeg.datasets.MNEDataset) but in my understanding both require to read data in memory first. 
So is there any function, that read this data interactively. I have seen other data class, but they are for particular dataset. I am talking about a generic dataset in this format

```
-data
--class0
---file1
---file2
--class1
---file1
---file2

```
The idea is inspired from pytorch ImageFolder class.
Thanks

tczhangzhi, 2023-04-25:
Yes, sorry for the late reply.

Today, we discussed your point. We agree that the dataset class based on the file tree structure is necessary and easy to use, but we still have the following considerations:
 
1. different file formats: Just as images include png, jpg, and other formats, file readers need to support .edf, .bdf, .gdf, .cnt, .egi, .mff, and other formats. We can refer to `mne` and other libs (https://mne.tools/stable/auto_tutorials/io/20_reading_eeg_data.html). Users may not need to configure, but the API needs to automatically check the file type to match the corresponding read functions.

2. meta-information: Unlike the complete representation of images, EEG signal files cannot exist independently. It requires a description file to specify meta-information, such as the names of electrodes corresponding to different channels, voltage values of the EEG, etc. Incremental args should be designed to allow users to specify corresponding information from APIs or files.

4. Segmentation: usually a recording file will include several seconds of EEG signals and several events. We should consider allowing users to set windows of EEG signal splits via parameters.

5. Stream: Recently, online learning and online inference have received extensive attention in brain-computer interfaces. In addition to static file structures, stream data (file) may also need to be considered. It probably shares the same underlying implementation as the file tree.

Overall, this requirement is feasible and necessary, but the interface design is still under consideration. Maybe I'll implement a quick draft implementation to get further insight, or you can start working on this if you want.

tczhangzhi, 2023-04-25:
Personally I considered the following architecture (but it's still under discussion because it's too aggressive):

`Class` (used to organize data structures and expose to `train_test_split` interface):
`EEGDataset` (interface)
|-- `WindowDataset` (interface)

`Mixin` (used to solve input and output problems):
`LMDBIO`: used to read and write EEG record cache
`StreamIO`: for reading and writing streaming data
`MNEIO`: for reading and writing EEG recording files
`NumpyIO`: for reading and writing preprocessed baseline datasets

The following basic implementations are provided:
`LMDBDataset` (implement WindowDataset) = `WindowDataset` mix `LMDBIO`
`StreamDataset` (implement WindowDataset) = `WindowDataset` mix `StreamIO`
`FileDataset` (implement WindowDataset) = `WindowDataset` mix `MNEIO`
`NumpyDataset` (implement WindowDataset) = `WindowDataset` mix `NumpyIO`

`RecordFolder` (implement EEGDataset) = `FileDataset` + `FileDataset` + `FileDataset`
`FeatureFolder` (implement EEGDataset) = `NumpyDatase`t + `NumpyDataset` + `NumpyDataset`/`LMDBDataset` + `LMDBDataset` +`LMDBDataset`

Refactor the following baseline dataset modules:
`DEAPDataset` (`FileDataset/LMDBDataset` multiple inheritance)
`DREAMERDataset` (`FileDataset/LMDBDataset` multiple inheritance)
`SEEDDataset` (`FileDataset/LMDBDataset` multiple inheritance)
`SEEDIVDataset` (`FileDataset/LMDBDataset` multiple inheritance)
`AMIGOSDataset` (`FileDataset/LMDBDataset` multiple inheritance)
`MAHNOBDataset` (`FileDataset/LMDBDataset` multiple inheritance)
etc.

talhaanwarch, 2023-04-25:
Hi, thanks for the response

In my understanding the solution of first three parts is allow user to create its own `read_file` function. And then pass this function to main dataset class (SEEDIVDataset as an example) as parameter. No matter what file user is reading, the user should define its own function, that read that file and convert it into standard mne object. Because there can be different types of files, and user want different processing on each file. So let user to do it by creating a function and pass the function to Dataset class and then to constructor function. 
In the `read_file` function, as user is creating a mne object, the user can also pass the meta files. For example names of electrodes etc.
Regarding segmentation, either we can define it in user `read_file` or create a new function for that. 

In short, instead of reading file transform_producer function, allow the user to write a reader function and pass the function to transform_producer.

The next part can be to read file names and labels from dataframe or csv similar to some thing built in keras flow_from_dataframe functionality. 

The other possibility to create a class for `transform_producer` and let user to inherit the function and modify this. but as it require a lot of refactoring, i would thing, the former approach is easier to implement.

tczhangzhi, 2023-04-26:
I agree that reading data from a `Dataframe` is also a good idea, I noticed that there are some datasets on Kaggle in `csv` format (https://www.kaggle.com/datasets/birdy654/eeg-brainwave-dataset-feeling-emotions).

I reserve my opinion on the manual handling of `read_file` by the user. This is because the user needs actually to manually customize too much content, or consider the applicability of the returned results to the following aspects:
1. Read the file (according to the file structure)
2. Divide the window and attach meta information (e.g., window start time)
3. Applying transforms
4. Match labels

However, I agree that a separate `transform_producer ` is a good solution, and the user can override the `read_file` method and not care about other methods. It already looks pretty mild compared to my solution above. 😄

talhaanwarch, 2023-04-30:
I just opened a PR #22 to show what I am thinking

closed 2023-05-18 by tczhangzhi