torcheeg/torcheeg #23: Read metadata from csv/dataframe

talhaanwarch, 2023-05-09:
Hi, I want to bring your attention toward a mostly used case scenario i.e read file_name and labels from pandas dataframe or a csv. 
Here is expected input.
```
pd.DataFrame([
 {'path': 'data/groupa/Sub1Post.mat', 'label': 1},
 {'path': 'data/groupa/Sub1Pre.mat', 'label': 1},
 {'path': 'data/groupa/Sub2Post.mat', 'label': 1},
 {'path': 'data/groupb/Sub1Post.mat', 'label': 0},
 {'path': 'data/groupb/Sub1Pre.mat', 'label': 0},
 {'path': 'data/groupb/Sub2Pre.mat', 'label': 0}])
```
As compared to directory reader suggested in #22 , reading metadata from dataframe provides more flexibility at user end. 
For example, i can do simple classification, but if i have to do complex processing after reading data using dataframe approach i can do that. My particular use case is to subtract pre from post and then fed it to cnn. Regardless of usecase, dataframe meta data approach allow user to build complex input functions.

Expected DataReader
```
dataset = DFDataset(io_path=f'raw_output', 
                    root_path='data',
                    num_worker=12, 
                    read_func =   read_file,  
                    dataframe = df,
                    online_transform=transforms.Compose([
                          transforms.ToTensor()    
                      ]),  
                    label_transform=transforms.Compose([
                      transforms.Select('label'),
                      ])
                     
                      )
    ```

tczhangzhi, 2023-05-13:
Thanks for your insight, but I think #22  is a better choice, it reduces the cost for user understanding and provides an out-of-box API.

In this (advanced) case #23, it may be a better choice for the user to manually process the data reading and hand it over to MNEDataset. Because I don't see what extra work it helps the user to deal with (the user still needs to define read_func).

talhaanwarch, 2023-05-14:
Regarding [MNE dataset ](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.MNEDataset.html#torcheeg.datasets.MNEDataset)my understanding might be wrong. But I think it need to bring the data into memory first. For example the data i have is around 100GB, but ram in my system is 32 GB, so i cant to convert the whole data into MNEDataset at once. So I need to a way to iteratively read and process the data

tczhangzhi, 2023-05-15:
That's a good point, let me think about it.

talhaanwarch, 2023-05-17:
If seems good, I can start working on it.

tczhangzhi, 2023-05-18:
I think it works, but we are using `loky` to refactor the dataset module recently. You can complete this idea, and I will refactor it after merging your pull requests.

tczhangzhi, 2023-08-22:
Since the PR has been successfully merged, I'll be closing this issue.

closed 2023-08-22 by tczhangzhi