torcheeg/torcheeg #39: Could not reproduce the accuracy of ViT

amrta-coder, 2023-06-25:
Is there anybody reproduced the accuracy of ARJUN et al’S VIT on DEAP datasets?

In the related paper of <<Introducing attention mechanism for EEG signals: Emotion recognition with vision transformers>>,
the author claimed that the ViT proposed could deal with raw data of EEG and achieve an accuracy of 98%~99%.
However, I could not reproduce it. I am wondering about the performance of ViT in dealing with EEG.
In my experiment, the ViT is hard to train due to the lack of data.

leehexx, 2024-07-18:
i also meet this issue, a lot of results can not be reproduced. i wonder if there exist some tricks in experiments of EEG

wfan-os, 2024-09-13:
I could only reach the accuracy of 55%. It seems like it is impossible to reach 98%.

leehexx, 2024-09-16:
probabaly there exsits data leakage problem in this experiment.

wfan-os, 2024-09-25:
> probabaly there exsits data leakage problem in this experiment.

I think it's impossible, because the train data didn't contain the emotion label.

Student-xtLin, 2024-11-11:
> i also meet this issue, a lot of results can not be reproduced. i wonder if there exist some tricks in experiments of EEG

I meet this issue as well! Many model results are either not reproducible or show significant discrepancies upon reproduction. I suspect it might be a data-related issue. For instance, when I use the data from the eeg_feature_smooth folder in the SEED-IV dataset, the model generally performs well ( only in GroupbyTrial experiment， CrossSubject experiment can not be reproduced） . However, when I switch to using data from the eeg_raw_data folder along with DE feature extraction in torcheeg, the model's performance drops significantly, almost to the point of being ineffective. Have you experienced anything similar?