torcheeg/torcheeg #40: Preprocessing of mahnob data sets

evercy, 2023-07-29:
Hello, if you want to use mahnob data set for end-to-end feature learning from raw data, is it not possible for this project (torcheeg) to directly use raw data as input, but also to pre-process (filtering, artifact removal, etc.)?

tczhangzhi, 2023-08-22:
You have the flexibility to accomplish a variety of tasks by combining different transforms. Below are some use-case scenarios that might interest you:

Use raw data, apply BaselineRemoval and MeanStdNormalize, then reshape into a 2D tensor (number of electrodes x data points). Here's how:

```
dataset = MAHNOBDataset(
    io_path=f'./example_mahnob_eegnet/mahnob',
    root_path='./Sessions',
    offline_transform=transforms.Compose([
        transforms.BaselineRemoval(),
        transforms.MeanStdNormalize(),
        transforms.To2d()
    ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select('feltVlnc'),
        transforms.Binary(5.0)
    ]),
    chunk_size=128,
    baseline_chunk_size=128,
    num_baseline=30,
    num_worker=4
)
```

Use raw data, decompose it into different frequency bands, and then apply BaselineRemoval.

```
dataset = MAHNOBDataset(
    io_path=f'./example_mahnob_fbcnet/mahnob',
    root_path='./Sessions',
    offline_transform=transforms.BandSignal(apply_to_baseline=True),
    online_transform=transforms.Compose([
        transforms.BaselineRemoval(),
        transforms.ToTensor()
    ]),
    label_transform=transforms.Compose([
        transforms.Select('feltVlnc'),
        transforms.Binary(5.0)
    ]),
    chunk_size=128,
    baseline_chunk_size=128,
    num_baseline=30,
    num_worker=4
)
```

Use differential entropy as a feature and map it onto a 9x9 grid.

```
dataset = MAHNOBDataset(
    io_path=f'./example_mahnob_ccnn/mahnob',
    root_path='./Sessions',
    offline_transform=transforms.Compose([
        transforms.BandDifferentialEntropy(sampling_rate=128, apply_to_baseline=True),
        transforms.BaselineRemoval(),
        transforms.ToGrid(MAHNOB_CHANNEL_LOCATION_DICT)
    ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select('feltVlnc'),
        transforms.Binary(5.0)
    ]),
    chunk_size=128,
    baseline_chunk_size=128,
    num_baseline=30,
    num_worker=4
)
```

Map raw data onto a 9x9 grid.

```
dataset = MAHNOBDataset(
    io_path=f'./example_mahnob_stnet/mahnob',
    root_path='./Sessions',
    offline_transform=transforms.Compose([
        transforms.BaselineRemoval(),
        transforms.MeanStdNormalize(),
        transforms.ToGrid(MAHNOB_CHANNEL_LOCATION_DICT)
    ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select('feltVlnc'),
        transforms.Binary(5.0)
    ]),
    chunk_size=128,
    baseline_chunk_size=128,
    num_baseline=30,
    num_worker=4
)
```

If these options don't meet your needs, you can always consult the [documentation](https://torcheeg.readthedocs.io/en/latest/auto_examples/examples_transforms.html) to customize your transforms until you achieve your desired output:

```
eeg = np.random.randn(32, 128)
transformed_eeg = transforms.BandDifferentialEntropy()(eeg=eeg)['eeg']
print(transformed_eeg.shape)
```

evercy, 2023-08-22:
Thanks a lot!

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi