torcheeg/torcheeg #42: Dreamer Dataset - not returning whole length trial (chunk_size = -1)

MicheleBarsotti, 2023-08-07:
### First of all Thanks for torcheeg. 

I am using the current last release
(v1.0.11 - Latest - on Dec 13, 2022)

I've noted a "bug" (or I think so) in the dreamer loader.
`

### Bug description
If you want to load the entire trial length 
 you pass  `chunk_size=-1` but **not all trials are loaded**.
```
from torcheeg.datasets import DREAMERDataset`
dataset = DREAMERDataset(
    io_path='./tmp_out/dreamer',
    mat_path='./tmp_in/dreamer/DREAMER_orginal.mat',    
    chunk_size=-1,
    num_worker=1)

for data, metadata in dataset:
    print(f"{metadata['subject_id']} - {metadata['trial_id']} - shape {data.shape}")
```
output:
> 0 - 0 - shape (14, 25472)
> 0 - 2 - shape (14, 25472)
> 0 - 7 - shape (14, 25472)
> 0 - 12 - shape (14, 25472)
> 0 - 14 - shape (14, 25472)
> 0 - 16 - shape (14, 25472)
> 1 - 0 - shape (14, 25472)
> 1 - 2 - shape (14, 25472)
> 1 - 7 - shape (14, 25472)
> 1 - 12 - shape (14, 25472)
> 1 - 14 - shape (14, 25472)
> 1 - 16 - shape (14, 25472)
> etc...

### Where is the bug
This bug seems inside the `transform_producer` function:
`from torcheeg.datasets.functional.emotion_recognition.dreamer import dreamer_constructor, transform_producer`
In fact, the passed value of `chunk_size` is overwritten if <=0 and not restored anymore in the loop.
line 60:
```
 if chunk_size <= 0:
     chunk_size = trial_samples.shape[1] - start_at
 # chunk with chunk size
 end_at = chunk_size
 # calculate moving step
 step = chunk_size - overlap
```

### Possible solution:
It would be sufficient to store the value of original `chunk_size` before the 
```
chunk_size_requested = chunk_size # store chunk_size variable
# loop for each trial
for trial_id in range(trial_len):
    chunk_size = chunk_size_requested # assign chunk size that will be changed later on the loop
```
(I did so in my data-exploration)
### Current Code Version :    
I've explored the github repo
And I have noted that what was inside the function `transform_producer` has been moved inside the main class DREAMERDataset

However,the current behaviour (not returning all trials with entire lenght)  is not fixed

Inside the trials-loop the chunk_size variable is modified and not restored at the beginning of the loop
https://github.com/torcheeg/torcheeg/blob/b6def3ab295fc9df9fc5dce495634e639def398b/torcheeg/datasets/module/emotion_recognition/dreamer.py#L188-L192
https://github.com/torcheeg/torcheeg/blob/b6def3ab295fc9df9fc5dce495634e639def398b/torcheeg/datasets/module/emotion_recognition/dreamer.py#L223-L230

Thanks again

tczhangzhi, 2023-08-08:
Indeed, you're right! We shouldn't modify the parameters passed by the user within the loop; instead, we should create a new variable. Otherwise, in the next iteration, it won't reflect the original value passed by the user (e.g., -1).

Would you like to create a PR? If not, I'll go ahead and address this issue and review other datasets.

In any case, thank you very much for identifying and bringing this to our attention!

MicheleBarsotti, 2023-08-10:
Thanks for the quick reply.
 I've just created the PR.

Thanks again

tczhangzhi, 2023-08-22:
Since the PR has been successfully merged, I'll be closing this issue. If you have any further questions or concerns, feel free to reopen it. Thank you for your contribution!