torcheeg/torcheeg #44: example_seed_feature_ada.py  is wrong.    Window system has multi-proces bug

shuentang, 2023-08-11:
File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 120, in spawn_main
    exitcode = _main(fd, parent_sentinel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 129, in _main
    prepare(preparation_data)
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 240, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 291, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen runpy>", line 291, in run_path
  File "<frozen runpy>", line 98, in _run_module_code
  File "<frozen runpy>", line 88, in _run_code
  File "d:\Personal\221236\Desktop\SEED\examples_seed_feature_ada.py", line 208, in <module>
    trainer.fit(source_loader, target_loader, test_loader, num_epochs=5)
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\site-packages\torcheeg\trainers\domain_adaption\ada.py", line 302, in fit
    **zip_loader = zip(source_loader, cycle(target_loader)) if len(**
                                    ^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\site-packages\torch\utils\data\dataloader.py", line 441, in __iter__
    return self._get_iterator()
           ^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\site-packages\torch\utils\data\dataloader.py", line 388, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\site-packages\torch\utils\data\dataloader.py", line 1042, in __init__
    w.start()
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\process.py", line 121, in start
    self._popen = self._Popen(self)
                  ^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\context.py", line 336, in _Popen
    return Popen(process_obj)
           ^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\popen_spawn_win32.py", line 45, in __init__
    prep_data = spawn.get_preparation_data(process_obj._name)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 158, in get_preparation_data
    _check_not_importing_main()
  File "D:\ProgramData\anaconda3\envs\pytorch\Lib\multiprocessing\spawn.py", line 138, in _check_not_importing_main
    raise RuntimeError('''
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.

shuentang, 2023-08-11:
I recently used pytorch for emotion classification on the seed dataset on windows.  Currently, I get bugs when I run examples_seed_feature_ada.py. 
I did some googling and it looks like torch is causing a multithreading bug when persisting sequences. I hope you can help to answer this question. thank you

tczhangzhi, 2023-08-22:
Sorry for the late reply.

Earlier versions did encounter parallelization issues on some Windows environments. As a quick fix, you can set the num_worker parameter to 0 in the dataset settings.

We've addressed this in the latest version, TorchEEG v1.1.0. For improved performance, you can upgrade to this version and follow our new example guide here: [TorchEEG Seed Domain Adaptation Example](https://torcheeg.readthedocs.io/en/latest/auto_examples/examples_seed_domain_adaption.html#sphx-glr-auto-examples-examples-seed-domain-adaption-py).

Thank you for bringing this to our attention!

closed 2023-08-22 by shuentang