torcheeg/torcheeg #53: Error while reading SeedDataset

flehn, 2023-08-29:
A few weeks ago I had no issues, however suddenly it is not possible to create a dataset with SeedDataset() Class anymore. 
It results in the ValueError: invalid literal for int() with base 10: ''
Please find attached a screenshot of the full error message.

I think it is related to the new parallel method. 

[image]

tczhangzhi, 2023-08-29:
The error you've encountered appears to be related to the recent version update. I can suggest two solutions to address this:

Update to the latest version, TorchEEGv1.1.0. However, make sure to delete the old dataset cache using the command:

```
rm -rf non_augmented_train
```

If you prefer, you can also revert to the older version, TorchEEGv1.0.11, by running:

```
pip install torcheeg==1.0.11
```

If you're working on a research project and want to ensure reproducibility, you can specify the version in your requirements.txt as:

```
torcheeg==1.0.11
# or torcheeg==1.1.0
```

In fact, we highly recommend using torcheeg1.1.0. Not only does it enhance dataset loading speeds and ensure cross-platform compatibility, but it also comes with built-in contrastive learning trainers, such as: [SimCLRTrainer](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.trainers.SimCLRTrainer.html#torcheeg.trainers.SimCLRTrainer).

Thank you for your support.

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi

SHAOHUASONGgit, 2024-08-03:
> The error you've encountered appears to be related to the recent version update. I can suggest two solutions to address this:
>
> Update to the latest version, TorchEEGv1.1.0. However, make sure to [...]

Well I got this error when I delete processed data and rerun my script, and it seems that the script get a None when getting sub id. So I think the script may read in a hidden doc or file but treat it as a record, then I ls the doc and delete .ipynb_checkpoints and rerun, error disappears :)