torcheeg/torcheeg #58: ImportError: cannot import name 'KFoldGroupbyTrial' from 'torcheeg.datasets'

xicheng105, 2023-10-08:
I ran the first example in the tutorial. It prompts the following warning:

2023-10-08 21:12:38,330 DEBUG MainThread matplotlib matplotlib data path: /home/louxc/anaconda3/envs/Tutorial/lib/python3.11/site-packages/matplotlib/mpl-data
2023-10-08 21:12:38,334 DEBUG MainThread matplotlib CONFIGDIR=/home/louxc/.config/matplotlib
2023-10-08 21:12:38,335 DEBUG MainThread matplotlib interactive is False
2023-10-08 21:12:38,336 DEBUG MainThread matplotlib platform is linux
2023-10-08 21:12:38,458 DEBUG MainThread matplotlib CACHEDIR=/home/louxc/.cache/matplotlib
2023-10-08 21:12:38,460 DEBUG MainThread matplotlib.font_manager Using fontManager instance from /home/louxc/.cache/matplotlib/fontlist-v330.json

Then the code in the second step reported an error:

ImportError: cannot import name 'KFoldGroupbyTrial' from 'torcheeg.datasets'.

xicheng105, 2023-10-08:
I know what's wrong. 'KFoldGroupbyTrial' is in 'torcheeg.model_selection', not in 'torcheeg.datasets'. You made a mistake.

dawin2015, 2023-10-08:
Do you mean here as below pic?

[image]

xicheng105, 2023-10-08:
Yes. Have you run examples in the tutorial yourself? I hope you guys can run them because the second example also has issues. for example, after run

import torch
from torcheeg.utils import plot_2d_tensor

img = plot_2d_tensor(torch.tensor(dataset[0][0]))

I get:

TypeError: Invalid shape (4, 9, 9) for image data

I'm very grateful for the work you do. But I want to say that if there are many coding errors in the tutorial, it is difficult not to doubt the quality of this toolkit. Sorry for being too blunt, but I hope this toolkit can be recognized by more people and continue improving.

dawin2015, 2023-10-08:
Thanks for your quick reply. I have the same faith with you to improve this repo. And what's more, if you can, just create a fork to fixed the error you have ccounted. I think this is a good way to support the repo's author. I am just a passby who just interests EEG. Sorry for being too blunt, too. :)

tczhangzhi, 2023-10-08:
Thank you for your kind response! Your revisions are entirely reasonable and I would like to merge it.

dawin2015, 2023-10-08:
> Yes. Have you run examples in the tutorial yourself? I hope you guys can run them because the second example also has issues. for example, after run
>
> import torch from torcheeg.utils import [...]

For your question, I think you use the wrong fuction because `plot_2d_tensor()` only accept two dimension array as input. If you use three dimension as para, I suggest you using the `plot_3d_tensor()`. FYI: [test_visualize.py](https://github1s.com/torcheeg/torcheeg/blob/HEAD/test/utils/test_visualize.py#L39)

tczhangzhi, 2023-10-08:
"I know what's wrong. 'KFoldGroupbyTrial' is in 'torcheeg.model_selection', not in 'torcheeg.datasets'. You made a mistake."

Thank you for reporting this error! We had previously fixed this error, but our CI went down. Thanks for bringing it to our attention.

We highly encourage any form of error reporting and are always ready to provide prompt assistance! https://github.com/torcheeg/torcheeg/commit/2ccfd740b16eefdfd004443b4f2d00cf6aeb1566

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi