torcheeg/torcheeg #61: Data mismatch EEGNet

rmib200, 2023-10-17:
I've been loading the BNCI2014001 from the moabb example, but when trying to make it run with the EEGNet, TSCeption, and CCNN there is always a shape mismatch that I can't fix. It seems to me that the way the data is being processed is by putting the channels last, but the networks need the channel first (after batch dim). 
For reference, I am applying a bandDifferentialEntropy, KfoldGroupbyTrial before training:

```
import torcheeg.datasets.moabb as moabb_dataset
from torcheeg import transforms

from moabb.datasets import BNCI2014001
from moabb.paradigms import LeftRightImagery

dataset = BNCI2014001()
dataset.subject_list = [1, 2, 3]
paradigm = LeftRightImagery()
dataset = moabb_dataset.MOABBDataset(
    dataset=dataset,
    paradigm=paradigm,
    io_path='./moabb',
    offline_transform=transforms.Compose([transforms.BandDifferentialEntropy()
                                          ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([transforms.Select('label')]))
    
    from torcheeg.model_selection import KFoldGroupbyTrial

k_fold = KFoldGroupbyTrial(n_splits=10,
                           split_path='./tmp_out/moabb/split',
                           shuffle=True,
                           random_state=42)
    
import torch
from torcheeg.models.cnn import TSCeption

from torch.utils.data import DataLoader

from torcheeg.trainers import ClassifierTrainer

import pytorch_lightning as pl

for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    train_loader = DataLoader(train_dataset, batch_size=64, shuffle=True)
    # for t in train_loader:
      # print("===", t[0])
    val_loader = DataLoader(val_dataset, batch_size=64, shuffle=False)

    model = TSCeption(num_classes=2,
                  num_electrodes=28,
                  sampling_rate=128,
                  num_T=15,
                  num_S=15,
                  hid_channels=32,
                  dropout=0.5)

    trainer = ClassifierTrainer(model=model,
                                num_classes=2,
                                lr=1e-4,
                                weight_decay=1e-4,
                                accelerator="gpu")
    trainer.fit(train_loader,
                val_loader,
                max_epochs=50,
                default_root_dir=f'./tmp_out/examples_pipeline/model/{i}',
                callbacks=[pl.callbacks.ModelCheckpoint(save_last=True)],
                enable_progress_bar=True,
                enable_model_summary=True,
                limit_val_batches=0.0)
    score = trainer.test(val_loader,
                         enable_progress_bar=True,
                         enable_model_summary=True)[0]
    print(f'Fold {i} test accuracy: {score["test_accuracy"]:.4f}')
```

The error:
`RuntimeError: Given groups=1, weight of size [15, 1, 1, 64], expected input[1, 64, 22, 4] to have 1 channels, but got 64 channels instead`

In this example, my dataset has a 1 in the first dim, and the batch (64) in the second dim. The electrodes count (22) follow but this order is not what the model expects. Looks like the model is expecting the batch dim last. 

This is the error with the EEGNet:

```
for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    train_loader = DataLoader(train_dataset, batch_size=64, shuffle=True)
    # for t in train_loader:
      # print("===", t[0])
    val_loader = DataLoader(val_dataset, batch_size=64, shuffle=False)

    model = EEGNet(chunk_size=128,
               num_electrodes=22,
               dropout=0.5,
               kernel_1=64,
               kernel_2=16,
               F1=8,
               F2=16,
               D=2,
               num_classes=2)
```
               
`RuntimeError: Given groups=1, weight of size [8, 1, 1, 64], expected input[1, 64, 22, 4] to have 1 channels, but got 64 channels instead`

With CCNN:

```
for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    train_loader = DataLoader(train_dataset, batch_size=16, shuffle=True)
    print("===", train_dataset)
    val_loader = DataLoader(val_dataset, batch_size=16, shuffle=False)

    model = CCNN(num_classes=2, in_channels=22, grid_size=(9, 9))

    trainer = ClassifierTrainer(model=model,
                                num_classes=2,
                                lr=1e-4,
                                weight_decay=1e-4,
                                accelerator="gpu")
```

`RuntimeError: Given groups=1, weight of size [64, 22, 4, 4], expected input[1, 16, 25, 7] to have 22 channels, but got 16 channels instead`
Can someone please point out how to properly load the dataset or change the model inputs?

tczhangzhi, 2023-10-17:
For the model to function correctly, you'll need to apply the appropriate preprocessing.

Please refer to the documentation specific to the model for guidance on this. For instance, in the [documentation](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.models.TSCeption.html#torcheeg.models.TSCeption) for TScpetion, it's mentioned that it can be used in conjunction with `To2d` to transform a single sample shape into `[1, electrode number, time point number]`:

```
dataset = moabb_dataset.MOABBDataset(
    dataset=dataset,
    paradigm=paradigm,
    io_path='./moabb',
    offline_transform=transforms.Compose([transforms.To2d()]), # see here!!!
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select('label'),
        transforms.Mapping({
            'left_hand': 0,
            'right_hand': 1,
        })
    ]))
```
Additionally, you'll need to tweak some hyperparameters within the model. Ensure parameters like `num_electrodes` align with your actual dataset:
```
model = TSCeption(num_classes=2,
                  num_electrodes=22,  # see here!!!
                  sampling_rate=128,
                  num_T=15,
                  num_S=15,
                  hid_channels=32,
                  dropout=0.5)
```
Here's a practical example for your perusal: [Colab Notebook](https://colab.research.google.com/drive/1Hp0i1aJHe9KQ3RfMwHMh7VFg5Bt01Cy2?usp=sharing).

rmib200, 2023-10-24:
Thank you for the answer. I noticed that the BNCI2014001 dataset contains 4 classes (left_hand, right_hand, feet, tongue) according to the moabb description on the website [http://moabb.neurotechx.com/docs/generated/moabb.datasets.BNCI2014_001.html#moabb.datasets.BNCI2014_001](http://moabb.neurotechx.com/docs/generated/moabb.datasets.BNCI2014_001.html#moabb.datasets.BNCI2014_001) but your example only uses 2 classes. While exploring the info.csv that the above code generates there are only labels for right and left hand. I've tried modifying the code to this:

```
dataset = BNCI2014001()
dataset.subject_list = [1, 2, 3]
paradigm = LeftRightImagery()
dataset = moabb_dataset.MOABBDataset(
    dataset=dataset,
    paradigm=paradigm,
    io_path='./moabb',
    offline_transform=transforms.Compose([
                                          transforms.To2d()
                                          ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
    transforms.Select('label'),
    transforms.Mapping({
        'left_hand': 0,
        'right_hand': 1,
        'feet': 2, 
        'tongue': 3
    })
  ]))
```

But still, only the right and left-hand classes are being downloaded. Do you know what could be happening or how to download the full dataset?

If I download the data using only moabb I can see the 4 classes, image for reference:
[image]

bruAristimunha, 2023-10-31:
Hi @rmib200,

Did my answer solve your question?

rmib200, 2023-10-31:
It did. Thank you very much Bruno

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi