torcheeg/torcheeg #63: The Moabb class stopped working

rmib200, 2023-10-23:
I'm using the following code to download the BNCI2014001 from moabb but is not working anymore. Last week it was fetching and processing the data correctly.

```
import torcheeg.datasets.moabb as moabb_dataset
from torcheeg import transforms

from moabb.datasets import BNCI2014001
from moabb.paradigms import LeftRightImagery

dataset = BNCI2014001()
dataset.subject_list = [1, 2, 3]
paradigm = LeftRightImagery()
dataset = moabb_dataset.MOABBDataset(
    dataset=dataset,
    paradigm=paradigm,
    io_path='./moabb',
    offline_transform=transforms.Compose([
                                          transforms.To2d()
                                          ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
    transforms.Select('label'),
    transforms.Mapping({
        'left_hand': 0,
        'right_hand': 1,
    })
  ]))
```

I'm attaching the error log for better comprehension. I'm not 100% sure if this is a moabb or a TorchEEG problem.

```
To use the get_shape_from_baseconcar, InputShapeSetterEEG, BraindecodeDatasetLoaderyou need to install `braindecode`.`pip install braindecode` or Please refer to `[https://braindecode.org`](https://braindecode.org%60/).
WARNING:moabb.utils:BNCI2014001 has been renamed to BNCI2014_001. BNCI2014001 will be removed in version 1.1.
WARNING:moabb.datasets.base:The dataset class name 'BNCI2014001' must be an abbreviation of its code 'BNCI2014-001'. See moabb.datasets.base.is_abbrev for more information.
/usr/local/lib/python3.10/dist-packages/moabb/datasets/download.py:55: RuntimeWarning: Setting non-standard config type: "MNE_DATASETS_BNCI_PATH"
  set_config(key, get_config("MNE_DATA"))
Downloading data from 'http://bnci-horizon-2020.eu/database/data-sets/001-2014/A01T.mat' to file '/content/io/moabb/raw/MNE-bnci-data/database/data-sets/001-2014/A01T.mat'.
dataset does not exist at path ./moabb, generating files to path...
MNE_DATA is not already configured. It will be set to default location in the home directory - /root/mne_data
All datasets will be downloaded to this location, if anything is already downloaded, please move manually to this location
/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py:1061: InsecureRequestWarning: Unverified HTTPS request is being made to host 'lampx.tugraz.at'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
100%|█████████████████████████████████████| 42.8M/42.8M [00:00<00:00, 34.7GB/s]
SHA256 hash of downloaded file: 054f02e70cf9c4ada1517e9b9864f45407939c1062c6793516585c6f511d0325
Use this value as the 'known_hash' argument of 'pooch.retrieve' to ensure that the file hasn't changed if it is downloaded again in the future.
Downloading data from 'http://bnci-horizon-2020.eu/database/data-sets/001-2014/A01E.mat' to file '/content/io/moabb/raw/MNE-bnci-data/database/data-sets/001-2014/A01E.mat'.
/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py:1061: InsecureRequestWarning: Unverified HTTPS request is being made to host 'lampx.tugraz.at'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
100%|█████████████████████████████████████| 43.8M/43.8M [00:00<00:00, 36.2GB/s]
SHA256 hash of downloaded file: 53d415f39c3d7b0c88b894d7b08d99bcdfe855ede63831d3691af1a45607fb62
Use this value as the 'known_hash' argument of 'pooch.retrieve' to ensure that the file hasn't changed if it is downloaded again in the future.
Downloading data from 'http://bnci-horizon-2020.eu/database/data-sets/001-2014/A02T.mat' to file '/content/io/moabb/raw/MNE-bnci-data/database/data-sets/001-2014/A02T.mat'.
/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py:1061: InsecureRequestWarning: Unverified HTTPS request is being made to host 'lampx.tugraz.at'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
100%|█████████████████████████████████████| 43.1M/43.1M [00:00<00:00, 37.0GB/s]
SHA256 hash of downloaded file: 5ddd5cb520b1692c3ba1363f48d98f58f0e46f3699ee50d749947950fc39db27
Use this value as the 'known_hash' argument of 'pooch.retrieve' to ensure that the file hasn't changed if it is downloaded again in the future.
Downloading data from 'http://bnci-horizon-2020.eu/database/data-sets/001-2014/A02E.mat' to file '/content/io/moabb/raw/MNE-bnci-data/database/data-sets/001-2014/A02E.mat'.
/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py:1061: InsecureRequestWarning: Unverified HTTPS request is being made to host 'lampx.tugraz.at'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
100%|█████████████████████████████████████| 44.2M/44.2M [00:00<00:00, 19.0GB/s]
SHA256 hash of downloaded file: d63c454005d3a9b41d8440629482e855afc823339bdd0b5721842a7ee9cc7b12
Use this value as the 'known_hash' argument of 'pooch.retrieve' to ensure that the file hasn't changed if it is downloaded again in the future.
Downloading data from 'http://bnci-horizon-2020.eu/database/data-sets/001-2014/A03T.mat' to file '/content/io/moabb/raw/MNE-bnci-data/database/data-sets/001-2014/A03T.mat'.
/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py:1061: InsecureRequestWarning: Unverified HTTPS request is being made to host 'lampx.tugraz.at'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/1.26.x/advanced-usage.html#ssl-warnings
  warnings.warn(
100%|█████████████████████████████████████| 44.1M/44.1M [00:00<00:00, 31.1GB/s]
SHA256 hash of downloaded file: 7e731ee8b681d5da6ecb11ae1d4e64b1653c7f15aad5d6b7620b25ce53141e80
Use this value as the 'known_hash' argument of 'pooch.retrieve' to ensure that the file hasn't changed if it is downloaded again in the future.
[... 41 lines cut ...]
     76                 # catch the exception

[/usr/local/lib/python3.10/dist-packages/torcheeg/datasets/module/base_dataset.py](https://localhost:8080/#) in __init__(self, io_path, io_size, io_mode, in_memory, num_worker, verbose, after_trial, after_session, after_subject, **kwargs)
     62                                               desc="[PROCESS]",
     63                                               total=len(records)):
---> 64                         self.save_record(io_path=self.io_path,
     65                                          io_size=self.io_size,
     66                                          io_mode=self.io_mode,

[/usr/local/lib/python3.10/dist-packages/torcheeg/datasets/module/base_dataset.py](https://localhost:8080/#) in save_record(io_path, io_size, io_mode, file, file_id, process_record, **kwargs)
    193                 # call process_record of the class
    194                 # get the current class name
--> 195                 obj = next(gen)
    196 
    197             except StopIteration:

[/usr/local/lib/python3.10/dist-packages/torcheeg/datasets/moabb/moabb_dataset.py](https://localhost:8080/#) in process_record(file, chunk_size, overlap, offline_transform, dataset, paradigm, before_trial, after_trial, **kwargs)
    126                 run_signal = before_trial(run_signal)
    127 
--> 128             proc = paradigm.process_raw(run_signal,
    129                                         dataset,
    130                                         return_epochs=False)

AttributeError: 'LeftRightImagery' object has no attribute 'process_raw'
```

bruAristimunha, 2023-10-23:
Hi Manuel Illanes,

Can you check the moabb version? And can you also check your torcheeg
version?

I suspect it will work with moabb 0.5 and not with moabb 1.0.

I had a major release today, and like every major release, it breaks some
things.

rmib200, 2023-10-23:
Yeah, definitely works with 0.5. I was looking at the moabb logs and realized there was a new stable version.

bruAristimunha, 2023-10-24:
nice @rmib200 ;)

tczhangzhi, 2023-11-09:
It looks like this issue has been resolved, so I'll close it. If you have other questions, feel free to reopen it.

closed 2023-11-09 by tczhangzhi