torcheeg/torcheeg #66: Loading SEED dataset

edugm94, 2023-11-09:
Hello,

I found this toolkit, which I find fascinating! I am trying to learn it, but I am having some problems to load the datasets. 
I am using Class `SEEDDataset` to load the dataset. However, it seems nothing is loaded. I attach a figure of the compiler: 
[image]

If someone could help me to understand what I am doing wrong, I would really appreciate it! 

Thanks in advance :)

tczhangzhi, 2023-11-09:
Your code looks fine. You may need to wait for the program to complete.

Providing more information will help locate your problem, or you can run this mini-case example showing how to load the SEED dataset:

https://colab.research.google.com/drive/1POEdPRKt3g2Nc4iihTKc_gKRHGSMPwjJ?usp=sharing

edugm94, 2023-11-09:
Hi, 

Thanks for fast response! 
I tried to do `print(dataset)` as it is done in the tutorial, and I got the error: `{TypeError}object of type 'NoneType' has no len()`

I understood that `root_path` should be the path to the files. In the case of SEED: .mat, labels.mat and readme.txt files. 
I have the latest version of torcheeg. 

Thanks so much for any insight!

tczhangzhi, 2023-11-09:
You may want to try the following code. I suspect there are two issues with the snippet of code you've provided:

1. 'io_path' should be a directory for caching processed results. The program will create this by default, so you just need to specify a directory that doesn't currently exist.
2. There seems to be an error with 'root_path'. It appears to be missing a '/' symbol.

```
dataset = SEEDDataset(io_path=f'./seed',
                      root_path=f'{path}/Preprocessed_EEG',
                      offline_transform=transforms.Compose([
                          transforms.BandDifferentialEntropy(sampling_rate=200),
                          transforms.ToGrid(SEED_CHANNEL_LOCATION_DICT)
                      ]),
                      online_transform=transforms.ToTensor(),
                      label_transform=transforms.Compose([
                          transforms.Select('emotion'),
                          transforms.Lambda(lambda x: x + 1)
                      ]),
                      chunk_size=200)
```

edugm94, 2023-11-09:
Hi! 

I had to remove the `chunk_size` argument, but now it worked :) 

Thanks so much!!!

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.

closed 2024-01-08 by tczhangzhi