torcheeg/torcheeg #70: Error: read_eeg()

edugm94, 2023-11-13:
Hello, 

I am running this code: 

```
dataset = SEEDDataset(io_path=f'./datasets_tmp/seed/RawDE',
                          root_path=path,
                          offline_transform=transforms.BandDifferentialEntropy(
                              band_dict={"delta": [1, 4], "theta": [4, 8], "alpha": [8, 14],
                                         "beta": [14, 31], "gmma": [31, 49]}),
                          online_transform=transforms.Compose([
                              transforms.ToTensor()
                          ]),
                          label_transform=transforms.Compose([
                              transforms.Select('emotion'),
                              transforms.Lambda(lambda x: x + 1)
                          ]),
                          num_worker=10,
                          verbose=True)
```

After creating the DataLoader, I try to do this: `next(iter(train_loader))` and I get the following error: 
```
File "/home/ego/miniconda3/envs/ldl/lib/python3.10/site-packages/torcheeg/io/eeg_signal.py", line 126, in read_eeg
    raise RuntimeError(
RuntimeError: Unable to index the EEG signal sample with key 10_20131204.mat_3204!
```

Every time I run it, I get a different key. It took a while to create the dataset, and the folder had all the corresponding recordings, so I do not know what I am doing wrong. 
Any help is appreciated!!

Thanks in advance.

tczhangzhi, 2023-11-13:
Your code looks good to me. Could you please provide more detailed information, such as the version of torcheeg you're using and the system you're running it on?

You might want to try running the following example to check for any errors: [Example Colab Notebook](https://colab.research.google.com/drive/1POEdPRKt3g2Nc4iihTKc_gKRHGSMPwjJ?usp=sharing).

I suspect that you might have forcibly stopped the program at some point, leading to a corruption in the io directory. You can try clearing the io with the following command:

```
rm -rf ./datasets_tmp/seed/RawDE
```

and then rerun the code.

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.

closed 2024-01-08 by tczhangzhi