torcheeg/torcheeg #71: Error into BandDifferentialEntropy implementation?

Nhix00, 2023-11-14:
I'm using torch EEG for my studies and i found that the differential entropy is calculated using this formula in BandDifferentialEntropy class:

```python
def opt(self, eeg: np.ndarray, **kwargs) -> np.ndarray:
        return 1 / 2 * np.log2(2 * np.pi * np.e * np.std(eeg))
``` 

Lokin into different papers and into the cited onces in the method, i discovered that the differential entropy feature is computed with variance, cause the sigma in the formula should be elevated at the power of 2, am I missing something or I'm right?

I think the correct computation shoul be:

```python
def opt(self, eeg: np.ndarray, **kwargs) -> np.ndarray:
        return 1 / 2 * np.log2(2 * np.pi * np.e * np.var(eeg))
``` 

or:

```python
def opt(self, eeg: np.ndarray, **kwargs) -> np.ndarray:
        return 1 / 2 * np.log2(2 * np.pi * np.e * (np.std(eeg))**2)
```

tczhangzhi, 2023-11-14:
We've reviewed the paper and the project, and your understanding is correct. This is indeed a serious bug. Thank you very much. If you're willing, you're welcome to create a pull request.

We will immediately investigate the cause of this error to prevent similar issues. I suspect it might have aligned with an incorrect implementation during the refactoring process.

tczhangzhi, 2023-11-14:
#72

closed 2024-01-03 by tczhangzhi