torcheeg/torcheeg #75: import ClassifierTrainer出现问题

2638289491, 2023-11-29:
运行实例代码中，出现ImportError: cannot import name 'ClassifierTrainer' from 'torcheeg.trainers' (D:\anaconda\lib\site-packages\torcheeg\trainers\__init__.py)错误，已检查anaconda文件夹中，两个函数都没有问题

tczhangzhi, 2023-11-30:
Could you check the version of torcheeg in the conda environment you are using? The version 1.1.0 of torcheeg works well.

Please check according to the following example:
https://colab.research.google.com/drive/1rdrwENcI0KQJZIuAJ6xlMZbyTvxfSwgd?usp=sharing

2638289491, 2023-12-01:
thank you，I can run this code successfully，but no results atfer running.
[image]

tczhangzhi, 2023-12-01:
It works well in all tests. Can you print the length of the dataset, or examine the loop of k_fold.split?

One approach is to clear the generated IO and rerun the program.

tczhangzhi, 2023-12-03:
Please also refer to this public kernel on Kaggle:

https://www.kaggle.com/code/szuzhangzhi/eeg-based-emotion-recognition/

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.

closed 2024-01-08 by tczhangzhi