torcheeg/torcheeg #77: DEAP dataset not found

VikasChidananda, 2023-12-12:
Hi, I have been trying out the "complete DL workflow" tutorial. I am unable to load the dataset, and I get this error

```Python
{
	"name": "FileNotFoundError",
	"message": "[Errno 2] No such file or directory: './data_preprocessed_python'",
	"stack": "---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In[3], line 7
      2 from torcheeg import transforms
      4 from torcheeg.datasets.constants.emotion_recognition.deap import \\
      5     DEAP_CHANNEL_LOCATION_DICT
----> 7 dataset = DEAPDataset(
      8     io_path=f'./examples_pipeline/deap',
      9     root_path='./data_preprocessed_python',
     10     offline_transform=transforms.Compose([
     11         transforms.BandDifferentialEntropy(apply_to_baseline=True),
     12         transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT, apply_to_baseline=True)
     13     ]),
     14     online_transform=transforms.Compose(
     15         [transforms.BaselineRemoval(),
     16          transforms.ToTensor()]),
     17     label_transform=transforms.Compose([
     18         transforms.Select('valence'),
     19         transforms.Binary(5.0),
     20     ]),
     21     num_worker=4)

File ~/anaconda3/envs/sst/lib/python3.11/site-packages/torcheeg/datasets/module/emotion_recognition/deap.py:142, in DEAPDataset.__init__(self, root_path, chunk_size, overlap, num_channel, num_baseline, baseline_chunk_size, online_transform, offline_transform, label_transform, before_trial, after_trial, io_path, io_size, io_mode, num_worker, verbose, in_memory)
    124 def __init__(self,
    125              root_path: str = './data_preprocessed_python',
    126              chunk_size: int = 128,
   (...)
    140              verbose: bool = True,
    141              in_memory: bool = False):
--> 142     deap_constructor(root_path=root_path,
    143                      chunk_size=chunk_size,
    144                      overlap=overlap,
    145                      num_channel=num_channel,
    146                      num_baseline=num_baseline,
    147                      baseline_chunk_size=baseline_chunk_size,
    148                      before_trial=before_trial,
    149                      transform=offline_transform,
    150                      after_trial=after_trial,
    151                      io_path=io_path,
    152                      io_size=io_size,
    153                      io_mode=io_mode,
    154                      num_worker=num_worker,
    155                      verbose=verbose)
    156     super().__init__(io_path=io_path,
    157                      io_size=io_size,
    158                      io_mode=io_mode,
    159                      in_memory=in_memory)
    161     self.root_path = root_path

File ~/anaconda3/envs/sst/lib/python3.11/site-packages/torcheeg/datasets/functional/emotion_recognition/deap.py:165, in deap_constructor(root_path, chunk_size, overlap, num_channel, num_baseline, baseline_chunk_size, before_trial, transform, after_trial, io_path, io_size, io_mode, num_worker, verbose)
    162 eeg_io = EEGSignalIO(eeg_signal_io_path, io_size=io_size, io_mode=io_mode)
    164 # loop to access the dataset files
--> 165 file_list = os.listdir(root_path)
    167 if verbose:
    168     # show process bar
    169     pbar = tqdm(total=len(file_list))

FileNotFoundError: [Errno 2] No such file or directory: './data_preprocessed_python'"
}

```

Any tips how to fix this? 

Thanks!

tczhangzhi, 2023-12-12:
To download the DEAP dataset, visit the website: https://www.eecs.qmul.ac.uk/mmv/datasets/deap/download.html.

Since you need to sign a License, TorchEEG cannot automatically download it for you. After downloading, modify the root_path to point to the downloaded and unzipped folder data_preprocessed_python.

For more details, you can refer to the documentation:
https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.DEAPDataset.html#torcheeg.datasets.DEAPDataset.

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.

closed 2024-01-08 by tczhangzhi