torcheeg/torcheeg #81: Loading the deap dataset, there is a problem, showing: 'NoneType' object has no attribute 'iloc'

logicvanlyf, 2023-12-19:
My code:
from torcheeg.datasets import DEAPDataset
from torcheeg import transforms

from torcheeg.datasets.constants.emotion_recognition.deap import \
    DEAP_CHANNEL_LOCATION_DICT

dataset = DEAPDataset(
    io_path='./deap',
    root_path='./data_preprocessed_python',
    offline_transform=transforms.Compose([
        transforms.BandDifferentialEntropy(apply_to_baseline=True),
        transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT, apply_to_baseline=True)
    ]),
    online_transform=transforms.Compose([
        transforms.BaselineRemoval(),
        transforms.ToTensor()
    ]),
    label_transform=transforms.Compose([
        transforms.Select('valence'),
        transforms.Binary(5.0),
    ]),
    num_worker=8,
)

print(dataset[0])

After running:
Traceback (most recent call last):
  File "D:\Codes\eeg-torch-project\torch_cnn_test.py", line 25, in <module>
    print(dataset[0])
          ~~~~~~~^^^
  File "D:\anaconda3\envs\eeg_torch\Lib\site-packages\torcheeg\datasets\module\emotion_recognition\deap.py", line 267, in __getitem__
    info = self.read_info(index)
           ^^^^^^^^^^^^^^^^^^^^^
  File "D:\anaconda3\envs\eeg_torch\Lib\site-packages\torcheeg\datasets\module\base_dataset.py", line 403, in read_info
    return self.info.iloc[index].to_dict()
           ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'iloc'
dataset already exists at path ./deap, reading from path...

tczhangzhi, 2023-12-19:
It seems your './deap' is corrupted. Please delete this folder, or point it to a new folder (e.g., './rG2qXCxrTb'), and run the file again.

tczhangzhi, 2023-12-26:
In the new version, we have provided more user-friendly error messages: https://github.com/torcheeg/torcheeg/blob/develop/torcheeg/datasets/module/base_dataset.py#L133-L136

It seems that this issue has been inactive for a long time, so I will close this issue. If you need additional help, please feel free to reopen it.

closed 2023-12-26 by tczhangzhi