torcheeg/torcheeg #82: Documentation for PhaseLockingCorrelation transform shows BandSignal implementation

chris7neves, 2023-12-19:
The documentation page for the PhaseLockingCorrelation class shows the instantiation of a BandSignal object instead of a PhaseLockingCorrelation object (https://torcheeg.readthedocs.io/en/v1.1.0/generated/torcheeg.transforms.PhaseLockingCorrelation.html).

Included in this issue is a screenshot with the problem displayed. 

[image]

tczhangzhi, 2023-12-23:
Thank you for your report, you are right, the example should be modified to:
```
transform = PhaseLockingCorrelation()
transform(eeg=np.random.randn(32, 128))['eeg'].shape
>>> (1, 32, 32)
```
Would you be willing to create a pull request that modifies this [line](https://github.com/torcheeg/torcheeg/blob/2b617a3bbda19d807eb57487a87fbedea055aef5/torcheeg/transforms/numpy/correlation.py#L94)?
Otherwise, I will revise it in the next version.

tczhangzhi, 2023-12-26:
Fixed: https://github.com/torcheeg/torcheeg/blob/develop/torcheeg/transforms/numpy/correlation.py#L94

closed 2023-12-26 by tczhangzhi