torcheeg/torcheeg #83: Simultaneous label transformation DEAP dataset

rmib200, 2023-12-22:
I was reviewing the DEAP approach in the examples and was wondering if there is possibility to transform the Valence and the Arousal labels simultaneously during the loading
 ```
dataset = DEAPDataset(
    io_path=f'./tmp_out/examples_pipeline/deap',
    root_path=".\data_preprocessed_python",
    offline_transform=transforms.Compose([
        transforms.BandDifferentialEntropy(apply_to_baseline=True),
        transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT, apply_to_baseline=True)
    ]),
    online_transform=transforms.Compose(
        [transforms.BaselineRemoval(),
         transforms.ToTensor()]),
    label_transform=transforms.Compose([
        transforms.Select('valence'),
        transforms.Binary(5.0),
    ]),
    num_worker=4)    
```

Particularly during this block of code:
```
label_transform=transforms.Compose([
        transforms.Select('valence'),
        transforms.Binary(5.0),
    ]),
```

It takes a loot of time to process the signal before training, and since there are 2 important labels (valence, arousal), I wanted to process both simultaneously.
Any response is appreciated.

tczhangzhi, 2023-12-23:
Yes, you can use the following code:
```
dataset = DEAPDataset(
    io_path=f'./tmp_out/examples_pipeline/deap',
    root_path=".\data_preprocessed_python",
    offline_transform=transforms.Compose([
        transforms.BandDifferentialEntropy(apply_to_baseline=True),
        transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT, apply_to_baseline=True)
    ]),
    online_transform=transforms.Compose(
        [transforms.BaselineRemoval(),
         transforms.ToTensor()]),
    label_transform=transforms.Compose([
        transforms.Select(['valence', 'arousal']),
        transforms.Binary(5.0),
    ]),
    num_worker=4)    
```
The expected output of dataset[0] is:
```
tensor(...)
[1, 0] # label of valence and arousal
```
You can refer to the documentation: https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.transforms.Select.html#torcheeg.transforms.Select

rmib200, 2023-12-23:
Thank you very much, I am getting a better grasp of how the library works now.

rmib200, 2023-12-23:
Can the object returned by the Select class, as shown in the provided example, be used to train separate models for Valence and Arousal? Or is it designed to train a single model with combined Valence and Arousal labels?

tczhangzhi, 2023-12-24:
We conducted a variety of experiments. TorchEEG allows for both strategies you mentioned. For the first, you just need to modify https://github.com/torcheeg/torcheeg/blob/main/examples/examples_pipeline.py#L48. Use respectively:

```
label_transform=transforms.Compose([
        transforms.Select('valence'),
        transforms.Binary(5.0),
    ])
```

and

```
label_transform=transforms.Compose([
        transforms.Select('arousal'),
        transforms.Binary(5.0),
    ])
```

to initialize the dataset. For the second, we usually refer to it as multi-label classification. Predicting valence and arousal is often complementary, so thorough training and multi-label classification can help improve performance. Here is an example code.

tczhangzhi, 2023-12-24:
```
"""
Using TorchEEG to Complete a Deep Learning Workflow
===================================================

In this tutorial, we demonstrate a complete deep learning workflow using TorchEEG. We will cover the following aspects:

1. Utilizing Datasets and Transformers in TorchEEG
2. Data Partitioning Strategies in TorchEEG
3. Leveraging Models and Trainers in TorchEEG

"""

######################################################################
# Step 1: Initialize the Dataset
# ------------------------------
#
# We use the DEAP dataset supported by TorchEEG. Each EEG sample is set to
# be 1 second long, encompassing 128 data points. The baseline signal is 3
# seconds long, which we divide into three sections and then average to
# obtain the trial’s baseline signal.
#
# During offline preprocessing, we divide each electrode’s EEG signal into
# 4 sub-bands, calculate the differential entropy for each sub-band as a
# feature, perform debaselining, and map onto a grid. Finally, the
# preprocessed EEG signals are saved locally. For online processing, we
# convert all EEG signals into Tensors, making them suitable for neural
# network input.
#

from torcheeg.datasets import DEAPDataset
from torcheeg import transforms

from torcheeg.datasets.constants.emotion_recognition.deap import \
    DEAP_CHANNEL_LOCATION_DICT

dataset = DEAPDataset(
    io_path=f'../tmp_out/ban_dif_ent_bas_rem_to_int_gri/deap_1s',
    root_path='./data_preprocessed_python',
    offline_transform=transforms.Compose([
        transforms.BandDifferentialEntropy(apply_to_baseline=True),
        transforms.BaselineRemoval(),
        transforms.ToGrid(DEAP_CHANNEL_LOCATION_DICT, apply_to_baseline=True)
    ]),
    online_transform=transforms.ToTensor(),
    label_transform=transforms.Compose([
        transforms.Select(['valence', 'arousal']),
        transforms.Binary(5.0),
    ]),
    num_worker=8)

######################################################################
# Step 2: Split the Dataset into Training and Test Sets
# -----------------------------------------------------
#
# In this case, we use per-subject 5-fold cross-validation to split the
# dataset. During this process, we separate each subject’s EEG samples
# into training and test sets. We use 4 folds for training and 1 fold for
# testing.
#

from torcheeg.model_selection import KFoldGroupbyTrial

k_fold = KFoldGroupbyTrial(n_splits=10,
                           split_path='./examples_pipeline/split',
                           shuffle=True,
                           random_state=42)

######################################################################
# Step 3: Define the Model and Initiate Training
[... 106 lines cut ...]

for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    train_loader = DataLoader(train_dataset, batch_size=64, shuffle=True)
    val_loader = DataLoader(val_dataset, batch_size=64, shuffle=False)

    model = MultiLabelCCNN(num_classes=2, in_channels=4, grid_size=(9, 9))

    trainer = MultiLabelTrainer(model=model,
                                num_classes=2,
                                lr=1e-4,
                                weight_decay=1e-4,
                                accelerator="gpu")
    trainer.fit(train_loader,
                val_loader,
                max_epochs=50,
                default_root_dir=f'./examples_pipeline/model/{i}',
                callbacks=[pl.callbacks.ModelCheckpoint(save_last=True)],
                enable_progress_bar=True,
                enable_model_summary=True,
                limit_val_batches=0.0)
    score = trainer.test(val_loader,
                         enable_progress_bar=True,
                         enable_model_summary=True)[0]
    print(f'Fold {i} test accuracy: {score["test_accuracy"]:.4f}')
```

tczhangzhi, 2023-12-24:
In fact, there are a lot of experiments based on TorchEEG here. Unfortunately, we don't have enough manpower to add these experiments to the documentation and examples. If you are willing, you can create a pull request (at https://github.com/torcheeg/torcheeg/tree/main/examples) and introduce how to perform Multi-label classification. Otherwise, I might update the documentation in a few weeks.

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.

closed 2024-01-08 by tczhangzhi