torcheeg/torcheeg #84: Where is BCI2022Dataset? I cannot find it

garenasd945, 2023-12-25:
When I click into class BCI2022Dataset code , it says :

class BCI2022Dataset(BaseDataset):
    r'''
    The 2022 EMOTION_BCI competition aims at tackling the cross-subject emotion recognition challenge and provides participants with a batch of EEG data from 80 participants with known emotional state information. Participants are required to establish an EEG computing model with cross-individual emotion recognition ability. The subjects' EEG data were used for real-time emotion recognition. This class generates training samples and test samples according to the given parameters and caches the generated results in a unified input and output format (IO). The relevant information of the dataset is as follows:

    - Author: Please refer to the downloaded URL.
    - Year: 2022
    - Download URL: https://oneuro.cn/n/competitiondetail/2022_emotion_bci/doc0

but when I click the link "https://oneuro.cn/n/competitiondetail/2022_emotion_bci/doc0",the link is broken, where to find it ?

tczhangzhi, 2023-12-26:
The dataset download link is currently not working. For other detailed information, please refer to the official website:

https://www.worldrobotconference.com/cn/about/163.html

tczhangzhi, 2024-01-08:
Since there hasn't been any new reply for a long time, it seems like the issue has been resolved. I am going to close this issue, but if you need further help, please feel free to re-open it.