torcheeg/torcheeg #85: TypeError: LightningModule.toggle_optimizer() got an unexpected keyword argument 'optimizer_idx'

logicvanlyf, 2023-12-26:
When I run the example code in **torcheeg
/examples_deap_cwgangp.py**, the program reports an error on the penultimate line of code, which is this position: **trainer.fit (train_loader, val_loader, max_epochs = 1)**.

The specific error is as follows:

Traceback (most recent call last):
...
  File "D:\anaconda3\envs\eeg_torch\Lib\site-packages\torcheeg\trainers\generative\wgan_gp.py", line 441, in training_step
    self.toggle_optimizer(generator_optimizer, optimizer_idx=0)
TypeError: LightningModule.toggle_optimizer() got an unexpected keyword argument 'optimizer_idx'

How can I solve it?

tczhangzhi, 2023-12-26:
I noticed that `optimizer_idx` has been removed:
https://github.com/Lightning-AI/pytorch-lightning/commit/c2aa8c9e10936667dde573db3e0e597b40edaa52
You can use pytorch-lightning 1.9.0 or earlier versions.
```
pip install pytorch-lightning==1.9.0
```
You can also define a class to override the `training_step`:
```
class WGANGPTrainer(MyWGANGPTrainer):
    def training_step(self, batch: Tuple[torch.Tensor],
                          batch_idx: int) -> torch.Tensor:
        ...
        self.toggle_optimizer(generator_optimizer) # change according to the new API
        ...
```
Of course, you are also welcome to create a PR to fix this issue, if you would like. Otherwise, I will fix it in the next version.

tczhangzhi, 2023-12-28:
Fixed: https://github.com/torcheeg/torcheeg/commit/644ad29ed0331750b0c012704eed0e2012aa554b

closed 2023-12-28 by tczhangzhi