torcheeg/torcheeg #89: train_accuracy changes too drastically

garenasd945, 2024-01-10:
One confusion I have is that during training, during the execution of trainer.fit(train_loader, val_loader), the train_accuracy changes too drastically. For example, it drops directly from 1 to 0. I am not very clear about the principle. I used the DREAMER data set here, trained for 64 epochs, and implemented the binary classification of valence.

[image]

this is code：
`
from torcheeg.datasets import *
from torcheeg import transforms

from torcheeg.datasets.constants.emotion_recognition.deap import DEAP_CHANNEL_LOCATION_DICT

base_dir = './dreamer_DE'
path_io = base_dir + '/dataset_cache'
train_test_split_path = base_dir + "/train_test_split"
model_train_save_dir = base_dir + "/model"

num_classes = 2
num_electrodes = 14
lr = 1e-4
epochs = 64
DGCNN_features = 5   

from torcheeg.transforms import BandHiguchiFractalDimension
dataset = DREAMERDataset(io_path=path_io,
                         mat_path='/home/user21100322/dataset/EEG/DREAMER/DREAMER.mat',                              offline_transform=transforms.Compose([
                                #   transforms.BandDifferentialEntropy(),
                                  transforms.BandDifferentialEntropy(sampling_rate=128, band_dict={'delta': [0.5, 4], 'theta': [4, 8], 'alpha': [8, 14], 'beta': [ 14, 31], 'gamma': [31, 49]}, apply_to_baseline=False),  # 时间结构
                              ]),
                         online_transform=transforms.ToTensor(),
                         label_transform=transforms.Compose([
                             transforms.Select('valence'),
                             transforms.Binary(3.0),  # 五分制
                             ]),
                         num_worker=32)

print('*' * 100)
print(dataset[0][0])
print(dataset[0][0].shape)
"""
"""
from torcheeg.datasets import SEEDFeatureDataset
from torcheeg import transforms
from torcheeg.datasets.constants.emotion_recognition.seed import SEED_CHANNEL_LOCATION_DICT
import argparse
from torcheeg.transforms import Compose
from torch.utils.data import DataLoader
from torcheeg.models import CCNN
from torcheeg.models import DGCNN
from torcheeg.trainers import ClassifierTrainer
import pytorch_lightning as pl
from torcheeg.model_selection import train_test_split_groupby_trial
import torch

train_dataset, val_dataset = train_test_split_groupby_trial(dataset, split_path=train_test_split_path)
train_loader = DataLoader(train_dataset, batch_size=64, shuffle=False)
val_loader = DataLoader(val_dataset, batch_size=64, shuffle=False)

num_train = 1
accuracy_list = []
for i in range(num_train):
    model = DGCNN(in_channels=DGCNN_features, num_electrodes=num_electrodes, hid_channels=32, num_layers=2,num_classes=num_classes)
    trainer = ClassifierTrainer(model=model,
                                num_classes=num_classes,
                                lr=lr,
                                weight_decay=1e-4,
                                accelerator="gpu")
    trainer.fit(train_loader,
                val_loader,
                max_epochs=epochs,
                default_root_dir=model_train_save_dir,
                callbacks=[pl.callbacks.ModelCheckpoint(save_last=True)],
                enable_progress_bar=True,
                enable_model_summary=True,
                limit_val_batches=0.0)
    score = trainer.test(val_loader,
                        enable_progress_bar=True,
                        enable_model_summary=True)[0]
    print(f'test accuracy: {score["test_accuracy"]:.4f}')
    accuracy_list.append(score["test_accuracy"])
    pass

import numpy as np
accuracy_list = np.array(accuracy_list)
file_path = './accuracy_list_dreamer_DE.txt'
np.savetxt(file_path, accuracy_list, fmt='%.4f', delimiter='\t', header='Accuracy Data')
`

tczhangzhi, 2024-01-10:
The code below looks confusing to me, as it would likely result in a batch containing samples of the same class:
```
train_loader = DataLoader(train_dataset, batch_size=64, shuffle=False)
```
Could you try setting `shuffle` to `True` and see how it goes?

garenasd945, 2024-01-11:
[image]
I set the shuffle to True and the result get better, thank you~