torcheeg/torcheeg #100: 在最新版pytorch（2.2.0）安装的踩坑记录

INGg, 2024-03-27:
环境是新的conda环境，python版本为3.10，用的清华源

1. 首先从官网上下的torch直接安装
pip3 install torch torchvision torchaudio --index-url https://download.pytorch.org/whl/cu121

2. 然后文档写的是大于1.8都能安，然后直接复制命令
pip install torcheeg
然后报错：note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for spectrum
  Running setup.py clean for spectrum
Failed to build spectrum
ERROR: Could not build wheels for spectrum, which is required to install pyproject.toml-based projects
（上面还有一部分，说的是× python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [45 lines of output]）

3. 根据报错指令搜索安装：
pip install pyproject
pip install tokenizers
然后再次运行安装torcheeg指令，还是报一样的错误

4. 找到spectrum官方安装教程，pip安装不上spectrum，选择使用官方提供的conda安装方式
conda config --append channels conda-forge
conda install spectrum
成功安装spectrum库

5. 再次运行torcheeg安装命令就成功了

6. 正在测试examples/examples_dreamer_tsception.py文件会不会报错，后续结果会更新的

INGg, 2024-03-27:
该文件调到3个epoch跑完了没报错

INGg, 2024-03-27:
该文件调到3个epoch跑完了没报错

tczhangzhi, 2024-03-28:
This issue arises from the third-party package spectrum, and unfortunately, we are unable to resolve it on the torcheeg side. We need to wait for the resolution of https://github.com/cokelaer/spectrum/issues/83. However, upgrading your Python version could be beneficial. For example, to Python 3.12.2.

soumiadjellikh, 2025-03-22:
I have resolved this issue by installing the Microsoft Visual C++ Build Tools.