torcheeg/torcheeg #101: "grid_size" parameter not working in torcheeg.models.gan.BGenerator/BCGenerator

HarlockOfficial, 2024-03-28:
Hello, I was working on dataset augmentation using GANs on the Physionet MI dataset.

I instantiated the generator with the following parameters:
```
generator = BGenerator(in_channels=128, out_channels=1, grid_size=(58, 65))
```
I used this code in the train step:
```
x = torch.rand(batch_size, 128).to(device).to(torch.float32)
generated_data = generator(x)
```
The result was a torch tensor with shape``(batch_size, 1, 9, 9)`` correctly reflecting the `out_channels=1` but not using at all the `grid_size` parameter.

I checked the library source code for both networks (`BGenerator` and `BCGenerator`); the `grid size` parameter is only saved as a variable in the networks and does not affect any of the layer creations or their parameters.

The same error does not affect the related `Discriminators`; there, the function
```
@property
def feature_dim(self):
    ...
```
uses `self.grid_size`.

I want to ask if this error derives from my misuse of the library or if there is some change I should apply to fix it.

I hope the issue description is informative enough; otherwise, please let me know so that I can provide further details.

tczhangzhi, 2024-03-28:
Greetings,

`grid_size` refers to the spatial dimensions of the grid-like EEG representation. In your case, the grid-like EEG representation has the shape of `(1, 9, 9)`, so you need to set `grid_size=(9, 9)`.

The current code is working as expected. This is because the convolutional layer can be adapted to grids of any size, so for the `Conv2d` layer, we don't need any specific settings. However, differences in `grid_size` will cause changes in the input size of the linear layer. Therefore, we calculate the feature_dim through grid_size, for example, `nn.Linear(self.feature_dim, 1)`.

Feel free to ask any questions or pull requests.

HarlockOfficial, 2024-03-29:
Thanks for the fast reply.
I did not understand which parameters I can use to obtain an output with a shape like `(batch_size, 1, 58, 65)` instead of the current `(batch_size, 1, 9, 9)` or if every eeg has to be represented in a `(9, 9)` grid.

tczhangzhi, 2024-03-31:
Not every EEG has to be represented in a (9, 9) grid. It just corresponds to the grid below to represent the positional relationship between the electrodes of the EEG:
```
[['-', '-', '-', 'FP1', '-', 'FP2', '-', '-', '-'],
['-', '-', '-', 'AF3', '-', 'AF4', '-', '-', '-'],
['F7', '-', 'F3', '-', 'FZ', '-', 'F4', '-', 'F8'],
['-', 'FC5', '-', 'FC1', '-', 'FC2', '-', 'FC6', '-'],
['T7', '-', 'C3', '-', 'CZ', '-', 'C4', '-', 'T8'],
['-', 'CP5', '-', 'CP1', '-', 'CP2', '-', 'CP6', '-'],
['P7', '-', 'P3', '-', 'PZ', '-', 'P4', '-', 'P8'],
['-', '-', '-', 'PO3', '-', 'PO4', '-', '-', '-'],
['-', '-', '-', 'O1', 'OZ', 'O2', '-', '-', '-']]
```
I'm not sure `which parameters I can use to obtain an output with a shape like (batch_size, 1, 58, 65)` is trying to express. Please clarify your question. Generally speaking, for ground-truth EEG samples, you can consider using different [transforms](https://torcheeg.readthedocs.io/en/latest/torcheeg.transforms.html); for generated samples, you can consider using [EEGfuseNet](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.models.EEGfuseNet.html#torcheeg.models.EEGfuseNet) if spatial information is not considered.

closed 2024-03-31 by HarlockOfficial