torcheeg/torcheeg #115: Hann filter is not properly imported from SciPy for torch

schrievicode, 2024-06-07:
I was working through the [Deep Learning Workflow for TorchEEG](https://torcheeg.readthedocs.io/en/latest/auto_examples/examples_pipeline.html) and realized that it does not allow me to import transforms due to it attempting to import scipy's Hann filter from the wrong source.

`from torcheeg import transforms`
gave me:
ImportError: cannot import name 'hann' from 'scipy.signal' (_censored_\Lib\site-packages\scipy\signal\__init__.py)

I'm sorry, as a novice coder I'm not sufficiently familiar with pull requests yet so I simply thought to add the solution in here instead

I fixed it for myself by changing code in the following two files:
Lib\site-packages\torcheeg\transforms\numpy\band_pyeeg.py
Lib\site-packages\torcheeg\transforms\numpy\band.py

from:
`from scipy.signal import butter, hann, lfilter`

to:
```
from scipy.signal import butter, lfilter
from scipy.signal.windows import hann
```

tczhangzhi, 2024-06-10:
Thank you for your report. This issue is caused by a problem with the version of scipy. Someone has already created a pull request to address it. (#111)

Anyway, thanks for your kind contribution.

SHAOHUASONGgit, 2024-07-02:
SciPy 1.11.0 works well

closed 2026-10-17 by tczhangzhi