torcheeg/torcheeg #116: PermissionError

weg1989, 2024-06-12:
I am running torcheeg on anaconda on my windows 11.

from torcheeg.datasets import SEEDIVFeatureDataset
from torcheeg import transforms
from torcheeg.datasets.constants import SEED_IV_CHANNEL_LOCATION_DICT

dataset = SEEDIVFeatureDataset(root_path='E:\SEED\eeg_feature_smooth',
                               feature=['de_movingAve','psd_movingAve','dasm_movingAve','rasm_movingAve'],
                               offline_transform=transforms.ToGrid         (SEED_IV_CHANNEL_LOCATION_DICT),
                               online_transform=transforms.ToTensor(),
                               label_transform=transforms.Select('emotion'))
print(dataset[0])

running the above peice of code gives the following error

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File c:\Users\anaconda3\envs\eeg\lib\site-packages\torcheeg\datasets\module\base_dataset.py:60, in BaseDataset.__init__(self, io_path, io_size, io_mode, num_worker, verbose, after_trial, after_session, after_subject, **kwargs)
     [53](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:53)     for file_id, file in tqdm(enumerate(records),
     [54](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:54)                               disable=not self.verbose,
     [55](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:55)                               desc="[PROCESS]",
     [56](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:56)                               total=len(records),
     [57](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:57)                               position=0,
     [58](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:58)                               leave=None):
     [59](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:59)         worker_results.append(
---> [60](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:60)             self.save_record(io_path=self.io_path,
     [61](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:61)                              io_size=self.io_size,
     [62](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:62)                              io_mode=self.io_mode,
     [63](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:63)                              file=file,
     [64](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:64)                              file_id=file_id,
     [65](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:65)                              process_record=self.process_record,
     [66](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:66)                              verbose=self.verbose,
     [67](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:67)                              **kwargs))
     [68](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:68) except Exception as e:
     [69](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:69)     # shutil to delete the database

File c:\Users\anaconda3\envs\eeg\lib\site-packages\torcheeg\datasets\module\base_dataset.py:228, in BaseDataset.save_record(io_path, io_size, io_mode, file, file_id, process_record, verbose, **kwargs)
    [225](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:225) try:
    [226](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:226)     # call process_record of the class
    [227](file:///C:/Users/anaconda3/envs/eeg/lib/site-packages/torcheeg/datasets/module/base_dataset.py:227)     # get the current class name
...
--> [618](file:///C:/Users/anaconda3/envs/eeg/lib/shutil.py:618)         os.unlink(fullname)
    [619](file:///C:/Users/anaconda3/envs/eeg/lib/shutil.py:619)     except OSError:
    [620](file:///C:/Users/anaconda3/envs/eeg/lib/shutil.py:620)         onerror(os.unlink, fullname, sys.exc_info())

PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: '.torcheeg\\datasets_1718191934062_vU2Ie\\_record_0\\eeg\\data.mdb'

tczhangzhi, 2024-06-12:
I am trying to reproduce this issue, but I haven't been successful. Did you create the code on a disk without write permissions?

I plan to try other Windows servers, while I recommend using a Linux system to avoid potential errors.

weg1989, 2024-06-12:
I am trying this on my laptop, and haven't had any write permission issues in the past.

I am thinking the same, using Linux would be a better idea.

closed 2024-06-15 by weg1989