torcheeg/torcheeg #117: RuntimeError: mat1 and mat2 shapes cannot be multiplied in last Linear layer of TSCeption

Alexanderstaehle, 2024-06-12:
I would appreciate some help as I am running into an error which I can't figure out what I am doing wrong :/

I am using a custom dataset which loads .fif files and try to classify the data using the following code:
```
dataset = CSVFolderDataset(csv_path='data_format.csv',
                           offline_transform=transforms.Compose([transforms.To2d()]),
                           online_transform=transforms.ToTensor(),
                           label_transform=None,
                           num_worker=4)
k_fold = KFold(n_splits=5, shuffle=True, random_state=42)
results = []
for i, (train_dataset, val_dataset) in enumerate(k_fold.split(dataset)):
    train_loader = DataLoader(train_dataset, batch_size=64, shuffle=True)
    val_loader = DataLoader(val_dataset, batch_size=64, shuffle=False)
    model = TSCeption(num_classes=2, num_electrodes=3)
    trainer = ClassifierTrainer(model=model,
                                num_classes=2,
                                metrics=["accuracy", "f1score"])
    trainer.fit(train_loader,
                val_loader,
                max_epochs=50)
```
However, when fitting the model I get the following error. `RuntimeError: mat1 and mat2 shapes cannot be multiplied (960x2 and 15x32)` which appears to occur in the last Linear layer of the TSCeption model. I also tried the EEGNet but had the same issue.
The data looks fine imo (`dataset[0][0].shape = torch.Size([1, 3, 1000])` if that is correct, also the labels are read correctly i.e. binary 0 or 1).

If any other information is needed let me know!

tczhangzhi, 2024-06-12:
Thank you for your report. This issue was caused by a smaller odd number of num_electrodes. After applying a 1/2 num_electrodes convolution, the resulting feature map shape does not match the expected shape.

I added padding in the case of an odd number of num_electrodes to ensure it works correctly. You can test the following code:

```
import torch
import math
from torch import nn
from torcheeg.models import TSCeption

class MyTSCeption(TSCeption):
    def __init__(self,
                 num_electrodes: int = 28,
                 num_T: int = 15,
                 num_S: int = 15,
                 in_channels: int = 1,
                 hid_channels: int = 32,
                 num_classes: int = 2,
                 sampling_rate: int = 128,
                 dropout: float = 0.5):
        super(TSCeption, self).__init__()

        self.num_electrodes = num_electrodes
        self.num_T = num_T
        self.num_S = num_S
        self.in_channels = in_channels
        self.hid_channels = hid_channels
        self.num_classes = num_classes
        self.sampling_rate = sampling_rate
        self.dropout = dropout

        self.inception_window = [0.5, 0.25, 0.125]
        self.pool = 8

        self.Tception1 = self.conv_block(
            in_channels, num_T,
            (1, int(self.inception_window[0] * sampling_rate)), 1, self.pool)
        self.Tception2 = self.conv_block(
            in_channels, num_T,
            (1, int(self.inception_window[1] * sampling_rate)), 1, self.pool)
        self.Tception3 = self.conv_block(
            in_channels, num_T,
            (1, int(self.inception_window[2] * sampling_rate)), 1, self.pool)

        self.Sception1 = self.conv_block(num_T, num_S, (int(num_electrodes), 1),
                                         1, int(self.pool * 0.25))
        self.Sception2 = self.conv_block(num_T, num_S,
                                         (math.ceil(num_electrodes * 0.5), 1),
                                         (math.ceil(num_electrodes * 0.5), 1),
                                         int(self.pool * 0.25),
                                         padding=(0, 0, 1, 0) if num_electrodes % 2 == 1 else 0)
        self.fusion_layer = self.conv_block(num_S, num_S, (3, 1), 1, 4)
        self.BN_t = nn.BatchNorm2d(num_T)
        self.BN_s = nn.BatchNorm2d(num_S)
        self.BN_fusion = nn.BatchNorm2d(num_S)

        self.fc = nn.Sequential(nn.Linear(num_S, hid_channels), nn.ReLU(),
                                nn.Dropout(dropout),
                                nn.Linear(hid_channels, num_classes))

    def conv_block(self, in_channels: int, out_channels: int, kernel: int,
                   stride: int, pool_kernel: int, padding: int = 0) -> nn.Module:
        return nn.Sequential(
            nn.ZeroPad2d(padding) if padding != 0 else nn.Identity(),
            nn.Conv2d(in_channels=in_channels,
                      out_channels=out_channels,
                      kernel_size=kernel,
                      stride=stride), nn.LeakyReLU(),
            nn.AvgPool2d(kernel_size=(1, pool_kernel), stride=(1, pool_kernel)))
    
model = MyTSCeption(num_classes=2,
                  num_electrodes=3)
fake_eeg = torch.randn(2, 1, 3, 1000)

print(model(fake_eeg).shape)
```

tczhangzhi, 2024-06-12:
By the way, if it's convenient for you, you can create a pull request to fix this issue by modifying the code in the following file to match the code above: https://github.com/torcheeg/torcheeg/blob/main/torcheeg/models/cnn/tsception.py. This way, we can acknowledge your contribution.

Alternatively, I will fix it in the next version. Thank you for your bug report.

Alexanderstaehle, 2024-06-12:
Thanks for the answer! The fix works perfectly.

It is more your contribution than mine but I created a PR [here](https://github.com/torcheeg/torcheeg/pull/118) :)
I also noticed the same issue with EEGNet but trying to apply the same fix didn't seem to work, so maybe this is unrelated.

tczhangzhi, 2024-06-13:
Thanks for your PR, I will mention it in the release notes for the next version~

closed 2024-06-13 by tczhangzhi