torcheeg/torcheeg #119: The following error occurred in the seed dataset，examples/examples_seed_domain_adaption.py

HaveAnIdeaCy, 2024-06-13:
Many thanks to the contributors for open-source the project, which I was very interested in, providing a solid torch experiment flow.

The following error is reported in this example as to what is the specific situation of leaving an experiment. Use torch==1.11.0。

examples/examples_seed_domain_adaption.py

  File "C:\Users\13645\.conda\envs\eegt\lib\multiprocessing\reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
_pickle.PicklingError: Can't pickle <function <lambda> at 0x00000184280AD1F0>: attribute lookup <lambda> on __main__ failed
RuntimeError:
        An attempt has been made to start a new process before the
        current process has finished its bootstrapping phase.

        This probably means that you are not using fork to start your
        child processes and you have forgotten to use the proper idiom
        in the main module:

            if __name__ == '__main__':
                freeze_support()
                ...

        The "freeze_support()" line can be omitted if the program
        is not going to be frozen to produce an executable.
Epoch 0:   0%|          | 0/371 [00:00<?, ?it/s

tczhangzhi, 2024-06-13:
It seems that this issue is caused by a multi-processing error on the Windows system. Could you please set `num_workers` to 0 in the dataset instantiation to check if the error still persists? (BTW, I still recommend using Linux for deep learning experiments.)

HaveAnIdeaCy, 2024-06-14:
Thank you very much for your suggestion, I have already run the code. However, num_classes=2 should be changed to num_classes=3 on line 124.

But I still have doubts. This piece of code seems to be an experiment in terms of sessions and individuals, what should I do if I just need to cross individuals (15 experiments).

tczhangzhi, 2024-06-14:
>  num_classes=2 should be changed to num_classes=3

Yes, you are right. Could you please make a pull request?

> what should I do if I just need to cross individuals (15 experiments)

If you don't want to take a subset, e.g., session 1, to conduct experiments, you can just apply the leave-one-subject-out function on the whole dataset:

```
from torcheeg.model_selection import Subcategory

# subset = Subcategory(
#     criteria='session_id',
#     split_path=f'./examples_seed_domain_adaption/split/session')

######################################################################
# Step 4: Train the Model
# ----------------------------------------------
#
# For training, we make use of the ADA algorithm to adapt domain-invariant features across subjects.
#

from torch.utils.data import DataLoader
from torcheeg.model_selection import LeaveOneSubjectOut
from torcheeg.trainers import ADATrainer

scores = []
# for j, sub_dataset in enumerate(subset.split(dataset)):
# loo = LeaveOneSubjectOut(split_path=f'./examples_seed_domain_adaption/split/loo_{j}')

loo = LeaveOneSubjectOut(split_path=f'./examples_seed_domain_adaption/split/loo')

# for i, (train_dataset, test_dataset) in enumerate(loo.split(sub_dataset)):
for i, (train_dataset, test_dataset) in enumerate(loo.split(dataset)):

    extractor = Extractor(in_channels=5, num_classes=3)
    classifier = Classifier(in_channels=5, num_classes=3)

    ...
```

By the way, it is a more challenging setting.

HaveAnIdeaCy, 2024-06-14:
Yes, thank you very much for your help, I would be happy to pull up a request to fix this error.

How do I import my local data set if I need to add steps to data preprocessing?

tczhangzhi, 2024-06-14:
Thanks for your contribution!

> How do I import my local data set if I need to add steps to data preprocessing?

You can refer to the document to load the local dataset: https://torcheeg.readthedocs.io/en/latest/auto_examples/examples_datasets.html

```
import mne

def default_read_fn(file_path, **kwargs):
    # Load EEG file
    raw = mne.io.read_raw(file_path)
    # Convert raw to epochs
    epochs = mne.make_fixed_length_epochs(raw, duration=1)
    # Return EEG data
    return epochs

dataset = [CSVFolderDataset](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.datasets.CSVFolderDataset.html#torcheeg.datasets.CSVFolderDataset)(io_path='./csv_folder',
                           csv_path='./data.csv',
                           read_fn=default_read_fn,
                           online_transform=[transforms.ToTensor](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.transforms.ToTensor.html#torcheeg.transforms.ToTensor)(),
                           label_transform=[transforms.Select](https://torcheeg.readthedocs.io/en/latest/generated/torcheeg.transforms.Select.html#torcheeg.transforms.Select)('label'),
                           num_worker=4)
```

You can apply the data preprocessing steps you need in the `read_fn` at the file level. Or, you can use `online_transform` and `offline_transform` at the chunk level, which supports common transforms: https://torcheeg.readthedocs.io/en/latest/torcheeg.transforms.html

HaveAnIdeaCy, 2024-06-19:
Thank you very much for your advice. It's nice to use cross-individual experiments on datasets, but I'm still wondering, for example, after seed processing, does ‘_record_0‘ correspond to the first individual (subject) in the dataset, all of them?

If I need to draw some diagrams, such as signal display diagram, classification result diagram after prediction classification, feature diagram, etc., are there relevant drawing functions provided? Thank you very much for your help.