torcheeg/torcheeg #127: Test DGCNN model with SEED-IV's 'eeg_feature_smooth' (DE_LDS) and 'eeg_raw_data'，but the results vary greatly

Student-xtLin, 2024-10-28:
Hi~， I used SEEDIVFeatureDataset and SEEDIVDataset to test DGCNN model , but the results vary greatly , which SEEDIVFeatureDataset+deLDS converges quickly and reach a high accuracy(about 90%)  while SEEDIVDataset+transforms.BandDifferentialEntropyV1 slow and low accuracy(only 45%）.  Is this normal? The model settings are the same，only change the data.

code:
```
dataset = SEEDIVDataset(io_path=f'./examples_seediv_dgcnn/seediv',
                        io_mode='pickle',
                        root_path='../data/SEED-IV/eeg_raw_data',
                        offline_transform=transforms.Compose([
                            transforms.BandDifferentialEntropyV1(sampling_rate=200, band_dict=band_dict, apply_to_baseline=True)]),
                        online_transform=transforms.Compose([
                            transforms.ToTensor(),
                        ]),
                        label_transform=transforms.Select('emotion'))
          dataset = SEEDIVFeatureDataset(
                            io_path=f'./examples_seediv_dgcnn/seediv_feature',
                            io_mode='pickle',
                            root_path='../data/SEED-IV/eeg_feature_smooth',
                            feature=['de_LDS'],
                            online_transform=transforms.ToTensor(),
                            label_transform=transforms.Select('emotion'))
         ```

ZiyiTsang, 2025-04-09:
Hi I meet same problem. Seems the performance in raw data in SEEDIVDataset is hard to reproduce. Any new Update?

Student-xtLin, 2025-04-15:
Maybe due to Linear Dynamic Systems (LDS)? I used LibEER to reproduce and got a better result than torcheeg, getting around 72% for DGCNN in SEED-IV which used '../data/SEED-IV/eeg_raw_data' from SEED-IV and LDS by LibEER.