torcheeg/torcheeg #129: Subject Dependent Testing for SEED dataset

rexjz, 2024-11-30:
In the EEG emotion recognition task, both Subject Dependent Testing and Subject Independent Testing are typically required. However, the SEED series dataset in this project does not appear to provide an easy way for users to create dataset objects containing data from a specific subject.

rexjz, 2024-12-03:
After reading the documentation, I realized that using `Subcategory` makes getting data from different subjects easy.

```python
from torcheeg.model_selection import Subcategory
from torcheeg.datasets import SEEDDataset

seed_dataset = SEEDDataset(
  root_path=<SEED_preprocessed_eeg_folder>,
  offline_transform=None,
  online_transform=transforms.Compose([
    transforms.ToTensor(),
  ]),
  label_transform=transforms.Compose([
    transforms.Select('emotion'),
    transforms.Lambda(lambda x: x + 1)
  ])
)
cv = Subcategory(criteria='subject_id')
cv.split(seed_dataset)
```

closed 2024-12-03 by rexjz

tczhangzhi, 2024-12-03:
For subject dependent testing, k-fold cross-validation is typically performed on each individual subject, with performance reported either separately for each subject or as mean and variance across subjects. Therefore, it is recommended to refer to [KFoldPerSubject](https://torcheeg.readthedocs.io/en/latest/torcheeg.model_selection.html#kfoldpersubject).

For subject independent testing, experiments typically follow a leave-one-subject-out approach. It is recommended to refer to [LeaveOneSubjectOut](https://torcheeg.readthedocs.io/en/latest/torcheeg.model_selection.html#leaveonesubjectout).